TomWhitwell/TuringMachine #24: Turing Machine Mk II cv problems

The-Davers, 2016-05-12:
Hi,
Just finished building and I thought all was ok until I tried to use cv control. It doesn't seem to have any effect ie not working. Is there a fix?

Thonk-Steve, 2016-05-17:
Hi,

Sorry for slow response, so the CV input maps directly the knob where -5V = fully counter clockwise, 0v = midnight, +5v = fully clockwise.

So if you set the knob to midnight, then send it +5v the sequence should lock as if you turned the knob manually.

Steve

The-Davers, 2016-05-17:
Hi,
No problem, you must be busy. Yes I got it working it was just subtle. Thanks for the explanation though as now I know what it does. 

Kindest Regards
Dave

From: Thonk-Steve 
Sent: Tuesday, May 17, 2016 9:02 AM
To: TomWhitwell/TuringMachine 
Cc: The-Davers ; Author 
Subject: Re: [TomWhitwell/TuringMachine] Turing Machine Mk II cv problems (#24)

Hi,

Sorry for slow response, so the CV input maps directly the knob where -5V = fully counter clockwise, 0v = midnight, +5v = fully clockwise.

So if you set the knob to midnight, then send it +5v the sequence should lock as if you turned the knob manually.

Steve

—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2016-06-03 by TomWhitwell

chrisib, 2022-12-29:
Apologies for replying to this 6 years later, but this was the only official source I could find that details the CV input voltage.

I just want to check some assumptions based on what you said above:
- if the knob is fully clockwise, -5V would have the effect of centering the knob at midnight and any positive voltage would have no effect?
- similarly, if the knob is fully anticlockwise, +5V would have the effect of centering the knob and any negative voltage would have no effect?

TomWhitwell, 2022-12-29:
Yes that's right, but I think (without checking the schematic) that the specific +5v amount is not precise - might be 4,5,6v

chrisib, 2022-12-29:
Thanks for the speedy reply! That makes sense.  So if the knob is turned all the way in one direction the input voltage range is too small to rotate it all the way to the other extreme?

i.e. The only way for e.g. an input LFO to fully sweep the knob's range is to leave the knob somewhere near midnight and use a bipolar input to lock it one way or the other, rather than leaving the knob in a locked position and using a monopolar signal to rotate it all the way in the other direction?

TomWhitwell, 2022-12-29:
I think fully left and +10v would probably sweep it all the way