TomWhitwell/TuringMachine #29: Turing Machine won't power up.

stefanhelton, 2016-05-30:
Not sure what I may have done wrong, was hoping you could help me. After assembling the Turing Machine, I plugged it into a uZeus power module and nothing seemed to happen. When I sent a gate signal to the clock input and sent the out to an oscillator,  it triggered it in a continuous drone.  None of the LEDs are coming on, and none of the switches or potentiometers have any effect on the sound. Do you have any idea what my problem could be?

Thonk-Steve, 2016-05-30:
Best thing to start with is post some good photos of both sides of both boards.

stefanhelton, 2016-05-31:
Sorry if some of the board is hard to see, I'm having to do this from mobile and my camera isn't that great. Some of the solder points are a little sloppy, but as far as I can tell they all make solid contacts and don't bridge over to other solder points. I have access to a multimeter if that would help, I'm just not sure where to start checking.

stefanhelton, 2016-05-31:
Forgot the link to the pics, my bad

http://m.imgur.com/a/HUGIG

TomWhitwell, 2016-06-02:
The TL074 on the front board is upside down. 
Curiously, I did EXACTLY the same thing while building a module this weekend. I spotted it, threw away the TL074 (you could probably just turn it round), got a new one, and installed THAT upside down too. Eventually I managed to put it the right way around (!) and it worked perfectly.

closed 2016-06-02 by TomWhitwell