TomWhitwell/TuringMachine #30: Issue with sharf fin ramping on the output

ludotex, 2016-06-03:
My problem: 
It seems like the turing introduce some unwanted ramps of the main output at times. Sometimes it will even get stuck high, or low for a few seconds (see video)

Here is a video that shows the behavior. You can clearly see on the oscilloscope the ramps I am talking about. 

https://youtu.be/3ECbmk3Pn3U 

I have checked the resistances and ic positions and legs, and it looks good to me.
The only thing that I find weird is that when I measure the 470K Resistor (the one close to the diode) it reads 90K (maybe it's normal I dunno...).
I tried swapping the ICs that were the same but I still get the same problem.

Here is some pics of the build: 
[image]
[image]
[image]
[image]

Thanks for your help or advices on how to fix it.

TomWhitwell, 2016-06-03:
I think the ramps are normal; if you think about how it works, it fixes a binary number and shifts it; so shifting from 
01000000
00100000
00010000
00001000
will cause an increase in output - a ramp. 
I can't quite tell from the video, but I'm not sure if loops are locking correctly. 
If you test it with 8 step sequences it's easier to tell what's going on. 
With shorter step sequences, there is a higher probability that the sequence will be 
111
or 
000
- i.e. locking high or low. 
With 8 steps, the probability of it becoming 
11111111
or
00000000
is lower - just in terms of coin tosses.

ludotex, 2016-06-03:
Hello Tom,

Thank you for your quick reply. I will look at this in detail this weekend. 

Reading your explanation, it seems that it's working correctly after all then. I was expecting a more random signal without any ramps.

I think I can get a lock when the trimmer is almost fully counter clockwise.

closed 2016-06-03 by ludotex

TomWhitwell, 2016-06-03:
If you want a different kind of CV output, the Volts expander is a configurable DAC - so the ramping effect doesn't happen - you can define how the binary is converted into CV

ludotex, 2016-06-03:
Thanks. I will look into that sometimes.

Looking forward to do some music with the turing machine :)

Really enjoying your modules btw, have a radio music as well. I like the
different directions you take in your module concepts.