TomWhitwell/TuringMachine #31: Turing Machine still not working

stefanhelton, 2016-06-03:
Hi, I'm the one who had the backwards IC from thread #29, which got closed.  After flipping the TL074 like you suggested, the row of LEDs at the top now turn on and stay on when the module receives power, but nothing else is different as far as I can tell.  Any idea what may still be going on?

TomWhitwell, 2016-06-03:
Assuming you're putting  square wave clock into the clock (you said 'gate' earlier) it might be that you've fried the TL074 - I replaced mine when I did the same thing 
The TL074 handles the clock input, the noise circuit and the CV output, so it's fairly critical

stefanhelton, 2016-06-30:
No difference with a new TL074, unfortunately. Do you know of any other potential issues or anyone who repairs Turing Machine modules that you would recommend?

stefanhelton, 2016-06-30:
Also, it was a square wave rather than a gate, sorry for getting my words mixed up there

TomWhitwell, 2016-09-15:
Did you resolve this? Closing due to inactivity.

closed 2016-09-15 by TomWhitwell