TomWhitwell/TuringMachine #32: Turing Machine MKii Clocking Issue

olijbailey, 2016-06-05:
Help Tom,

Could I please ask your opinion on a clicking issue with the MKii TM? I seem to be able to get the lights moving using the end of rise pulse from my PEG but my usual clock sources don't seem to work, nor do the square LFOs I have (although the 258J VCO does work). For some reason, it seems to like a wonky wave. Any ideas if I can swap anything out to make it clockable from other sources?

Thonk-Steve, 2016-06-05:
Hi oli,

Please can you list the modules you're having problems with and also detail
what actually happens with them?

Steve

olijbailey, 2016-06-05:
Hi Steve,

Thanks for getting back to me. Well, I get no response from my Pamela's Workout or from the clock from my shuttle control. I tried the square wave from the shuttle control and that's no go either. Oddly though, it does clock from the EOR of a PEG, from the QPLFO and from the square wave of the 258J VCO, which is just weird. I'm going to try changing the shift register ICs and maybe look at the transistor in the noise section. I'm guessing I should be able to clock it from the Pam - right?

Cheers

Oli

TomWhitwell, 2016-06-05:
Hi, before looking at the shift register or transistor ICs, look at the clock input - bottom right hand corner on the first schematic. 
If one of the resistor values is wrong (swapped with somewhere else) in this area, that might explain why you were having trouble clocking. 
https://github.com/TomWhitwell/TuringMachine/blob/master/Collateral/PDF_Schematics%20/TuringMachine2_Schematic_May2016.pdf
Mine clocks with everything in my system - from raw Arduino outputs to all my oscillators to maths EOC pulses

Thonk-Steve, 2016-06-05:
Yeah I really think this is a build issue, I've been clocking with a Pamela
all the way through development. If you upload build pics it'll help us
chekc and rule stuff out.

olijbailey, 2016-06-06:
That's great - thank you both. I just needed to know that it should be clocking from those sources. 

Cheers

Oli

closed 2016-09-15 by TomWhitwell