TomWhitwell/TuringMachine #33: Trimming the Lock Question

soapdistant, 2016-06-16:
Hi! 

Not so much an issue, as i'm fairly sure my TM mkII is working, it lights up and makes all the right moves when i feed a clock into it.

My issue is, when i attempt to [trim the lock](https://vimeo.com/163160088), i've gone through the steps in the video but using a VCO from my Korg MS-20 Mini into the clock input, but the TM doesn't respond to the input. I get no indicator led on the clock in, the top led's remain constantly on, and I hear no output from the TM's out. 

Is this an indication that i may have messed up the build, or could it be that shouldn't be using an MS20-mini to generate the square wave? If so, is there another method i could use to trim the lock as i currently do not have a Eurorack VCO?

It's a great kit, was immense fun to build and i can't wait to start using it, 

Many thanks,

Soap

Thonk-Steve, 2016-06-16:
Hey Soap,

It may be that the MS20-mini audio output is just far too quiet and outputting at the level of regular gear rather than modular levels. 

Steve

soapdistant, 2016-06-16:
Hi Steve, thanks for the quick response! 

That's relieving to know i haven't messed up the build, presumably then i'll just need to amplify the output a bit? Would i be able to boost the signal enough using a Mackie Onyx 1220i Desk? Or would i need a specific line level to modular boosting option?

Many thanks again,

Soap

Thonk-Steve, 2016-06-16:
Yeah that may well work, there's normally enough headroom on Mackie mixers, if you slowly increase the gain you may see it start to work... if the signal gets distorted too much it could cause some weirdness also though.

soapdistant, 2016-06-16:
Thanks Steve, will give it a try!

closed 2016-09-15 by TomWhitwell