TomWhitwell/TuringMachine #34: trouble with turing machine mk ii

lundl, 2016-06-19:
hello, I have enjoyed building the turing machine today untill i got to the trimming portion and encountered a problem, the unit , when set up as the video instructs emits a popping/screaching sound and the leds(including the pulse led) flash, there is no way to trim this out and the problem is at its worst with the scale know all the way clockwise. i took it apart, double checked my work and even checked some of the tracers and resistors with a multi meter, little stumped and looking forward to using the module, thanks in advance. here is a pic of my build. http://imgur.com/uKbinxa

Thonk-Steve, 2016-06-20:
Can you post a picture of how you're patching the trimming process? What oscillator you're using etc?

pbizzlr, 2016-06-20:
this sounds similar to my own problem

lundl, 2016-06-20:
thanks for the quick response,  im using the synthesizer box's square wave output(pictured), i have a dixie 11 coming this week so i could try it with that 
[image]
[image]

lundl, 2016-06-22:
Looking to get this resolved, tried trimming with the Dixie as well and it was just as bad

Thonk-Steve, 2016-06-22:
have you tried trimming it by ear running as a sequencer?
It's just as effective it just takes a bit longer.

so patch a regular clock to the input, then patch the output to the 1v/oct input of your oscillator.
set the scale knob to max and set the length to 8.

Put the main knob at midnight then lock fully clockwise, the sequence should loop perfectly, if it doesn't keep dialling the trimmer counter clockwise gently until it does.

lundl, 2016-06-22:
I will try that tonight , anything else you would recommend?

lundl, 2016-06-23:
I have resolved the problem, i followed steve's instructions about trimming by ear. the problem i was having was not being patient with the process. im sorry for taking up your time steve, i will be buying more products from thonk, thanks steve.

closed 2016-06-23 by lundl

Thonk-Steve, 2016-06-23:
Hey, no worries, do you have any tips on how we could improve the process to avoid that?

lundl, 2016-06-23:
well i think i would find the sweet spot simply so i was pretty aggressive with the trimming. perhaps noting that if you have problems with the trimming trying it as a sequencer by ear is a good alternative? again, i think the problem was mostly mine. thanks again. btw, the shipping was amazing, england to western canada in four business days is insanely good.

Thonk-Steve, 2016-06-23:
yeah actually I think giving the sense of the kind of small angle of
deviation required could be helpful. thanks!