TomWhitwell/TuringMachine #35: Turing Machine mk ii no noise out

kybernaughty, 2016-06-20:
Hi Tom, 
First of all congratulations on this great module! I've never seen that level of documentation and parts labelling in a DIY project. It was a real charm to build! Now my TM2 works well enough, it clocks and locks and spits out sequences and pulses,  but the noise jack is silent. Now I've read up on this and imagine the TL074 might be at fault. I double checked my build, multimeter in hand, so I'm positively sure that the resistors next to the jacks are correct, and the solder joints, too, and of course the orientation of all ICs. Can you please help me out? 

Secondary mission. : I'm also in the dark as to what the CV input (?) is supposed to do - I get the impression it changes the clocking and pitch behavior of the output...

Thanks a lot in advance !

Best,
Yuru

Thonk-Steve, 2016-06-20:
If the module is working fine but the noise jack is silent then you must
have a simple issue with the output jack or components near it. The module
wouldn't work if the noise circuit wasn't working internally, so inspect
near that jack again and consider reflowing the joints there.

The CV input is simply CV control that does the same as the big knob. So
set the big knob to midnight' then put 5V into the CV input and it's the
equivalent of turning the knob fully clockwise... -5v for fully
anti-clockwise.

kybernaughty, 2016-06-21:
Hi Steve,
Thanks for your quick reply! Issue solved- in my attempt to lock the sequences I had also turned down the noise...I stumbled across a remark on muffs concerning what the trimmer does. Of course I also reflowed various joints & swapped the TL074 for good measure. 
Best,
Yuru

closed 2016-09-15 by TomWhitwell