TomWhitwell/TuringMachine #36: Trimmig and other problems

matthijskwak, 2016-06-26:
Hi Tom,

I recently bought an assembled Turing Machine and have some questions about it. 

For one, when following the trimming procedure in your Vimeo clip I only get a steady output when the probability knob is fully CCW. As a result, the response at noon and fully CW are not what they should be (i.e. the range is completely off). 

Second, the bits seem to slowly cluster when I lock the sequence. For instance, A sequence may start as 01011010 and loop but than it slowly morphs to 11110000 and loops like that without ever changing. 

I have contacted the person who I bought the Turing Machine from, but in the mean time I thought I'd ask you as well, hoping you might be able to help me out. 

Kind regards,

Matthijs

Thonk-Steve, 2016-06-26:
Hi Matthijs,

Best place to start is to upload photos of both sides of both PCBs so we can rule out any obvious build issues.

Cheers

Steve

matthijskwak, 2016-07-01:
Hi Steve, 

I returned the module and it appears it was a faulty TL074. 

Kind regards,

Matthijs

closed 2016-07-01 by matthijskwak