TomWhitwell/TuringMachine #39: diy osmosis

zaejin, 2016-07-02:
Hi, great fun building the kit but it seems there's a problem.
When it's attached to the power, it makes every socket in my modular output a clicky slow square tone.
Even on different modules. The other bits work okay when the turing isn't plugged in the power.

I'm using a synthrotek psu.
Cheers,
Victor.

Thonk-Steve, 2016-07-02:
I would start by using a multimeter to check if you have continuity on the
Turing power connector between -12v and Gnd, +12v and Gnd or +12v and -12v.

Sounds like you have a short circuit in the Turing build somewhere to me.

zaejin, 2016-07-02:
Hi thanks for your time.
There doesn't seem to be a short between those pins which I guess is where the whole of the circuit converges,
(i am a noob at this so I don't know where to look for shorts, those traces look awfully close).
I've had a look at all the solder joins and they seem ok.

Thonk-Steve, 2016-07-02:
Hi, do you mean you tested with a meter or visually? The short will present
itself at the header to the meter but physically could be anywhere on
either board. You can't do this test visually.

zaejin, 2016-07-02:
tested with a meter, yes. but I'm at passive mult level of skill here.

TomWhitwell, 2016-07-02:
I'd double check that all the chips are the right way round and in the right places

zaejin, 2016-07-02:
yeah they're checked to the build docs. all okay.

Thonk-Steve, 2016-07-04:
Ok I think best thing to do is take photos of both sides of both PCBs and
we can take it from there

zaejin, 2016-07-04:
Okay, taken some pictures as best I can without a tripod.
Many thanks for the effort.
[image]
[image]
[image]
[image]

swahananda, 2016-07-07:
It looks like your 1n4001 on the back board are both oriented in the same direction. This is no bueno as diodes only allow power to flow through one direction and your -12 rail is currently aiming back into your system. The one on the bottom left of your image needs to be reversed.

zaejin, 2016-07-07:
it works you f***ing wizard!! bit tricky resoldering the component but I managed it.
Thanks thanks thanks you efforts are appreciated thonk and friends.

Thonk-Steve, 2016-07-07:
Ah it was nice to come check up and find this :)

swahananda, 2016-07-07:
Oh heck yeah! Glad to be of some service! Now if only I can get my own Turings sorted, life will be even sweeter.

closed 2016-09-15 by TomWhitwell