TomWhitwell/TuringMachine #40: Turing machine assembled, no sound

relougarou, 2016-07-02:
Hi, first thanks for the very well documented project!
I'm experiencing an issue on the calibration test, since the module seems not to behave the way it should.

When I plug the system, with no cable plugged, all the top LEDs light up, but when I start the calibration test by plugging a square on clock entry, the out on my mixer, set the parameters correctly, i can't obtain anything with my screwdriver turn.
Nonetheless, some LEDs behave strangely with cables as I reset the module. 
I tried to take a video to illustrate the behavior of the module: https://youtu.be/6TmMlCM6wdw

Thanks in advance for your support.
Best!

swahananda, 2016-07-06:
Right off the bat, the clock led should be off until a clock signal is fed into it. If it ain't blinkin' then something is stinking'. Before one of the big guys asks, please upload some hi-res front and back photos of the board (preferably cleaned of all flux). Thanks

TomWhitwell, 2016-09-15:
Have you made any progress with this? Closing due to inactivity, please reopen if you're still having issues.

closed 2016-09-15 by TomWhitwell