TomWhitwell/TuringMachine #41: Pulses mk2 problem

fishman77, 2016-07-07:
Just finished building my Turing Machine mk2, Pulses mk2, and Volts mk2. The only problem I've run into is 2 of the outputs/LEDs are not working on the Pulses. Specifically the 1+2 and the 1+2 4+7 outs (top and bottom right side). Can you give me some pointers where I can test to see what's going on?  I went over the IC a second time and that fixed another problem. This was a first for surface mount components for me so there could be a soldering issue. Thanks!

Thonk-Steve, 2016-07-07:
Hey, do you have a digital multimeter with continuity testing?

fishman77, 2016-07-07:
Sent from The Moon

fishman77, 2016-07-07:
Yes I do.  
Adam

Sent from The Moon

Thonk-Steve, 2016-07-07:
actually you should start by trying another ribbon cable to rule that out and then looking at the soldering on the header on the main turing and the header on the Pulses before diving into the components.

fishman77, 2016-07-07:
I got it working. I inspected all the IC pins and headers and there were a couple maybes. It's works great now. Thanks for the help!

closed 2016-09-15 by TomWhitwell