TomWhitwell/TuringMachine #43: No sign of life.

jestern77, 2016-09-05:
Assembled the complete kit as suggested in the docs. Seemed pretty easy. I plugged the TM (power + clock) nothing worked. Realized I forgot one 100nF cap on the ic (back) board. Soldered. Plugged in, no life. Checked all transformers to be correct and IC to be the right ones. Redid all solder points, still nothing. Sometimes when I unplug the power the two leds (clock and first bit) light up. Changed the TL074 (just in case) no sign of life.

Attached some pictures. Any clue? Maybe the lacking 100nF could have screwed up some IC? strange though...

[image]
[image]
[image]
[image]

TomWhitwell, 2016-09-15:
Can't see anything unusual in the pictures - have you made any progress? Does the clock LED flash when you connect it to a clock?

jestern77, 2016-09-15:
HI Tom, no unfortunately doesn't work. The led doesn't lit up but the noise output does output noise. So the power goes there. Maybe some burned IC?

TomWhitwell, 2016-09-15:
The clock input is the same chip as the noise generator, so it's hard to see how that could be. The 100nf would definitely not be a problem. 
It's a mystery - are you able to try it in another case? Do you have a friend with a working Turing you could compare it with? 
You could certainly try replacing the ICs, but it's an unusual case.

jestern77, 2016-09-15:
hmm I refilled some of the holes, still no sign of life. Checked the LED orientation. I can send it to repair to a friend and see what went wrong...

closed 2017-12-19 by TomWhitwell