TomWhitwell/TuringMachine #44: Turing Machine mk2 error

paxxxx, 2016-09-14:
Hello guys! I just connect the turing machine mk2.

I put a square wave in clock , and the clock light its ok , same tempo than wave.
I get noise and pulse, but nothing comes from Out...i was following the calibrate video , and it seems that i should have sound from out isnt it?
Also... all the top lights are fixed all the time.

Anyone can help me on this?
thanks very much guys.
Picture attached.

[image]

TomWhitwell, 2016-09-14:
You won't get sound from Out, but voltages to drive an oscillator. 

Lights will be fixed on until you put a clock in and put the main knob at 12 O'clock.

paxxxx, 2016-09-14:
Thanks for your reply Tom! 

Now i connected the out to 1v Oct input of a synth, and then it sounds as loop...then i move the main knob at 12 o clock, and i see the lights moving , but after 2 or 3 compasses, all lights are gone and never come back!

Maybe im doing something wrong?? sorry if its a basic question.

paxxxx, 2016-09-14:
i mean, the clock still pulsing, but pulse output light are off now, and top lights are all off too.

paxxxx, 2016-09-15:
Nothing... The module accept the clock, and create some pulses and cvs, but after few seconds, all the top lights are off, the pulses lights are off, and dont create any cvs :(

TomWhitwell, 2016-09-15:
That sounds like a soldering error - somewhere you may have missed a pin, swapped a component or inserted a diode the wrong way around.

closed 2016-09-15 by TomWhitwell