TomWhitwell/TuringMachine #47: Pulse outputs Failing

123gravy, 2016-12-30:
Hey, 

 I have just put together the Turing machine plus volts and pulses expanders.  All working fine apart from the pulse output on the Turing machine itself and two outputs on the pulses expander (the '1+2' and '1+2+4+7'). There is no output from these at all. Has anyone experienced something like this? Or is there something I should try? Solder joints look ok

Thanks

TomWhitwell, 2016-12-30:
Check orientation of all the chips on the back PCB of the main Turing module. Try swapping the AND chips to see if that makes a difference

123gravy, 2016-12-30:
Hi Tom,

 Thanks for the reply.  Orientation of the chips are all good.  Which chips
are the AND chips?

Isn't it strange that both the 1 output and the 2 output are all good but
not the 1+2 output on the Pulses?

Cheers

Thonk-Steve, 2016-12-30:
Photos of the build would help, did you also double check the soldering of the expansion header on the main module? Do you have any other expander modules like Volts, Voltages etc?

TomWhitwell, 2016-12-30:
You can also try connecting pulses to the gates header. That will tell you if the problem  is in the main module, or the pulses module.

123gravy, 2017-01-04:
Hi,

 Thanks for the replies.  Please excuse my slow reply, I was away from my rig for a few days.

I do have another expander, Volts.  Seems to be working ok.  

The solder joints of the expansion headers look ok to me but I'll attach a few photos. 

I attached the Pulses module to the Gates header on the Turing Machine and the Pulse output on the Turing Machine then works, but still not the two (the '1+2' and '1+2+4+7') outputs on the Pulses module.  What does that suggest there must be an issue with both the TM and the Pulses builds?  Sorry to be a pain

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

123gravy, 2017-01-04:
I double checked all of the expander headers and re-soldered a couple of pins that looked as though the solder hadn't gone through to the other side quite as well when you pry up the black pin support on the header.  Still all the same

123gravy, 2017-01-04:
I've checked all of the jacks on the Pulses with a beep test.  All fine.  Re-soldered the IC as it was a bit rough.  Still the same.  I'm convinced that it is just the Pulses module.  I just can't suss out which component...
[image]
[image]
[image]
[image]
[image]

123gravy, 2017-01-05:
The Pulse output on the Turing Machine also works fine without any expanders attached

123gravy, 2017-01-10:
I should also mention that the outputs of the Pulses do not carry pitch but they do give a signal.  I've been using them to trigger drums which works fine.  However when I send the signal to an oscillator they each trigger in only one pitch.  Every output has the same pitch except for output 1 which has a higher repeated pitch, no variation following the turing machine...

TomWhitwell, 2017-01-10:
Pulses outputs are triggers - normally low (below 1v) becoming high (roughly 8v) when they are active. They're not expected to have any pitch content.

123gravy, 2017-01-10:
Ah ok. I only got that impression from this video https://youtu.be/Le26BIqB8Y8 about 14 mins in. I can't see the full patch though.  So that's that query sorted. Thanks. Can you make any sense of the rest. (The 2 outputs not working on the pulses and the pulse output not working on the Turing machine only when the pulses is plugged in)?

TomWhitwell, 2017-01-10:
Sounds like a soldering problem around the chip on the expander

123gravy, 2017-01-10:
Ok cheers. I did touch up some soldering around the chip and nothing changed. It still could be better though so I'll try removing it altogether and starting fresh. Thanks

123gravy, 2017-01-10:
[image]
Hello again Tom.  OK so i re-soldered it again and this time also lost outputs 4, 2+4, and 4+7.  I touched it up again and managed to get it back to just missing the original 1+2 and 1+2+4+7.  As you can see from this photo I've made a dogs dinner of the PCB.  You can see some copper next to the IC.
 I'm wondering maybe if I've knackered the IC? Or now possibly the PCB?  Or both.  Would it be worth getting a new IC do you think?  Or have I killed this one?  Sorry to be a pain in the arse

TomWhitwell, 2017-01-10:
Hi, - that doesn't look great - I can't really tell from the pictures, but it looks like there could be any number of shorts between the IC pins. If you have access to a hot air soldering gun, lots of flux and soldering braid, it might be possible to clean it up, but tricky...

123gravy, 2017-01-10:
I do have a bit of flux and braid.  Yeah to be honest I haven't really been using the flux.  I'll give it a proper go again using those.  So you wouldn't be too worried about the exposed copper on the PCB?

TomWhitwell, 2017-01-10:
If it's just exposed, that's probably OK so long as nothing is shorting onto it.

123gravy, 2017-01-10:
Ok, thanks

123gravy, 2017-02-01:
Hey Tom,

 Cleaned up the IC.  Re-soldered like a machine! Same drama.  Re-soldered all resistors.  All working like a dream, Turing machine, volts and pulses.  Must have been a bad contact thing with the resistors on the Pulses.  Awesome kits! How annoying is ametures must be.  Cheers

closed 2017-02-01 by 123gravy