TomWhitwell/TuringMachine #48: Scale Pot Main Module

paxxxx, 2017-01-03:
Hello guys. I have 2 Turing Machines with all the expanders.
 
I just read that the Scale Pot in the main module... should do Cv changes like the 2 knobs on the Voltages expanders...
But in my 2 units.. the scale pot it seems to do nothing..

When i change knob possition in Voltages expander, i change the pitch of the sequence inmediatly. But not in the Main module..

Maybe i have something wrong in both units? or its not the same function?

Thanks very much for your help, my favourite sequencer of all time! .
Alberto

TomWhitwell, 2017-01-03:
The scale pot is a simple attenuator on the CV output of the main module. If it isn't working, you must have some kind of short circuit on the pins of the potentiometer - although that's an unusual mistake, and very unusual to make it twice!

paxxxx, 2017-01-03:
yes.. they dont attenuate nothing... same Cv output all the time. 
Attenuator on the voltages works ok.

I will check the pins. 

thanks!

paxxxx, 2017-01-08:
Hello Tom... Just to let you know it was my fault, i was listening the volts expander output , thats why i didnt hear the scale changes in the main module.

Everything ok in both modules!!!

Thanks man 

Ribé
p. +34 636 14 63 61

closed 2017-01-08 by TomWhitwell