TomWhitwell/TuringMachine #49: Problems with Random and Noise

bertog, 2017-01-04:
Hi Tom,

First thanks for the beautiful kit and instruction. This is my first ever build so I'm very newbie for this type of thnigs. The module that I've build seems to work but I think there a couple of issue and I need some hint ...

1. I've use the square out from my Mother32 for the calibration. I'm not able to here any sound from the Turing machine only noise. I've tried to calibrate checking the LEDs.

2. After the calibration the module seems always in lock mode. When I move the Lock pot the sequence change, but it stay lock forever in every position. I've found another guy with a similar issue here on GitHub and I've check the 8pin IC but it seems ok ...

3. Seems that the noise out isn't working. I've reflush the jack and some components near the jack without success.

May can I have some help?

Cheers and thank you. Guido

bertog, 2017-01-04:
Sorry Tom .... I've made a terrible calibration because I've not follow well the video. Everything is workin as expected ... sorry :)

Guido

closed 2017-01-04 by bertog

jamesbvaughan, 2017-07-15:
@bertog I seem to be having the same issue you described. What exactly did you do to solve it?

bertog, 2017-07-15:
You need to properly calibrate the module following the video linked at the end of the build guide

jamesbvaughan, 2017-07-15:
I have followed the steps in the video, but I get a constant tone and I don't get any noise output no matter what the position of the trimmer pot is. You mentioned that your noise output wasn't working. What did you do to get it to work?