TomWhitwell/TuringMachine #52: Non Locking Issue

pbizzlr, 2017-02-12:
I have built 1 mark 1 and 2 mark 2s and i cannot get any of them to lock.

I have soldered a 1M resistor of this one im working on as id read this may be a fix but no luck.

so next I have been swapping out 3904s to see if i can get a better response on the noise circuit but after 10x 3904s this doesnt seem to be giving me locking behaviour either. 

heres a vid of me trying to calibrate as per the thonk video https://www.dropbox.com/s/nhq1hsd0olvfhi1/Video%2012-02-2017%2C%2021%2026%2049.mov?dl=0 

actually the sound i get from the output when clocking with a saw wave doesnt seem to resemble that shown in the calibration video, granted its a different VCO (mines is taken from the MI module tester) but it seems different. The tone doesnt go constant for me, just glitchy, or off altogether.

ive been going back to this module on and off for over a year now, help me Tom your my only hope

Thonk-Steve, 2017-02-13:
> 2017%2C%2021%2026%2049.mov?dl=0

Hi, did you buy the MKII from Thonk?

pbizzlr, 2017-02-13:
depends, will you help if i say i did? :) 

i did not get it from you nope

Thonk-Steve, 2017-02-13:
No it just makes it more complicated as we don't know what parts you used. Could you upload photos of both sides of both PCBS?

pbizzlr, 2017-02-13:
no bother thank you

[image]
[image]
[image]
[image]

raimund, 2017-02-20:
Did you flip the expander cables? I don't know how it should be, but if red is always to the right, and your cables are mirrored, something could be wrong here …