TomWhitwell/TuringMachine #53: BOM xls missing 47k resistor

whodevil, 2017-02-16:
Hello,

I'm looking over the BOM and the PCB, and it appears as though the mouser BOM that is in xls format is missing a 47k resistor.

Thanks!

Thonk-Steve, 2017-02-16:
Hi, which BOM?

whodevil, 2017-02-17:
Hello, this is the one I'm talking about /Collateral/BOM/Turing v4 Mouser Export.xls
I think it is also missing two 1n4001 diodes and two ferrite beads.

whodevil, 2017-02-17:
It seems like this BOM is for a different project than the Turing Machine MKii. Here is what I have found missing so far:
- 47k resistor
- 2.2k resistor (9x)
- 1n4001 diodes (2x)
- two ferrite beads
- two fuses
- 78L09 voltage regulator (2x)
- 6 pin two row female socket (2x)
- 10 pin two row female socket
- 16 pin IC socket (x5)
- 14 pin IC socket (x4)
- 8 pin IC socket
- .1uf ceramic capacitor (8x) - There are 5 in the BOM but the build requires 13
- 100n ceramic capacitor (5x)
- hex stilts (2x)
- IC CD4050 (2x)
- IC CD4081 (2x)

There are 3 TL072 but only one is required.
There is an extra 10k resistor.
There are three 2N3903 transistors but only one is required.

It would probably make more sense for me to fix it and create a pull request.

Thonk-Steve, 2017-02-17:
Ah ok, yeah use this one - https://docs.google.com/spreadsheets/d/1FUUejrNd6qe-PUmlLi0BAgiGKWX9DxQH3Jt5qFOxPAQ/edit?usp=sharing

whodevil, 2017-02-17:
@Thonk-Steve Thanks! I think that document should be linked in the README.md or something. At the very least remove the old BOMs from the repo.

TomWhitwell, 2017-02-19:
Thanks, have now updated with a link to the correct BOM from Google Docs

closed 2017-02-19 by TomWhitwell