TomWhitwell/TuringMachine #56: No sound at output, no action on Leds

zoth4355, 2017-03-20:
Bought the turing from Thonk (complete kit with volts and pulses). The unit worked for months, then I sold to a buddy. Worked fine until a couple of weeks ago. The unit stopped locking and the seq leds  stalled, clock input doesn't seem to move anything or when it does it hangs almost immediately. He brought it back to me to check it out. Tried clocking it from 3 different sources (ALM's Pamela's, 4MS QCD and an SSF Square Wave. clock LED seems to work (does flicker), pulses LED is always lit (?), noise output works, and I hear nothing from the output (found this while calibrating the unit). I'd appreciate some help resolving this issue. Thank you.

TomWhitwell, 2017-03-20:
If it was working but is now not working, could it have suffered some strange voltage incident; it's protected against reverse voltage, but could have been something else. Swapping the chips would be one way to find out, start with the TL074 on the front PCB - that handles the clock. Alternatively- could it be a cold solder joint somewhere?

zoth4355, 2017-03-20:
Swaped the TL074, did get some action for a couple of second then stopped. I tested the scale knob and it does work, but the sequnce hang hafter little time, odd. I'll go back to my solders, hopefully it's a cold joint. Thank you for the super fast reply Tom.

TomWhitwell, 2017-06-21:
Hi, did you make any progress with this?

zoth4355, 2017-06-21:
Hi Tom,
Thanks for asking,
I've tried every trick in the book, reflowing the solders, recalibration and swapping some IC's and other stuff, still unstable, sometime locked for 10-15 mins when lucky (this was tested without the expanders attached). I got frustrated and bought new pcb's from Steven and ordered new components from mouser. The original unit, I think, had a bad trace but I can't confirm, I just couldn't find what went wrong, why it got crazy after a few month of working properly. 
Sold the fresh new unit to a buddy. Still stable at this time. All IC's are TI went for quality parts, took more care than usual when manipulating the pcb's and advised my friend to be careful when connecting the pulses and voltages. Now he want's me to order and build a vactrol mix for him... 
Might it strain the Turing module to have so many expanders attached to it? Don't see why but might as well ask.
Thanks
Pat

closed 2017-12-19 by TomWhitwell