TomWhitwell/TuringMachine #57: Newly built Mkii not working properly, can't calibrate

aamohr, 2017-03-31:
The lights work, noise output seems ok, VCO through out intermittent with static, no calibration happening and then finally not much response at all.  Appreciate any input.

Here are pics of the board.  
[image]
[image]
[image]
[image]

TomWhitwell, 2017-06-21:
Hi, sorry to hear that, I can't see anything obvious wrong, did you make any more progress with this?

petermorck, 2017-11-07:
Hi,

@aamohr left the module to me to take a look at it. I have managed to do the calibration, but like many others I found there's an extremely narrow trimmer interval between full CCW and random notes. I mean the place where you get a regular output is juuuust before the CCW end. Surely some value in the noise amp is not suitable. The trimmer could probably easily have been a 100k, but I didn't try that since it did calibrate in the end. However, this is not the main issue.

Now I find that if I set it to 2 steps, one of 3 things happen once I turn the change-knob to the center and back to full CW:
1. All max output values and every step is a trigger.
2. All min values and every step has no trigger.
3. Alternating values that are ALWAYS the same 2, i.e. they're not 2 arbitrary voltages frozen in a cycle, but they always end up being the same 2 alternating votlages.

I also noticed that if I'm holding the write switch up or down and change a longer sequence, the entire output range "slides" as if the output moves with the average of the sum of the steps.

I suspect the two problems are related or have the same root issue. Any hints about what?

jstamant, 2018-03-22:
@petermorck I can confirm that your edit is correct, and that you've figured out the expected behaviour of the OUT jack.

The OUT jack just puts out a voltage generated by the DAC0800 circuit. The DAC circuit puts out a voltage that is the binary-weighted sum of the 8 LEDs, if that's technically correct to say. For more info, you can look up discrete DAC circuits, and the relevance of "binary-weighted" as it relates to DAC circuits. All that to say that a bit in the first LED will produce the least change in the OUT jack voltage (1/256th of the OUT jack's voltage range), while the last LED will produce the most change in the OUT jack voltage (~1/2 or 128/256ths of the OUT jack's voltage range.

As for the trim pot value. I can't comment too much on that. I'd have to look into the noise circuit further to see if simply changing the trim pot value would broaden the range for calibration. If a simple change of values won't broaden the range, then you'd be able to do it by adding a resistor in series with the trim pot.