TomWhitwell/TuringMachine #59: No noise output + extremely tight range for randomizing sequence

Elberstein, 2017-05-29:
Hi,
I encounter some problems with this module. I double checked the position of the components and went back on some solders just to be sure, it looks a bit messy maybe now... but at first, the solders were shiny and quite clean for a first time :) 
There's no noise output from the jack but I think the module works because it generates some random sequences, but only when the lock pot is set to midnight. Every LED's are ok. It receives a clock from Maths or Grids or from a square VCO (I tried a VCO from my Pro2 and from Software via an ES-8). But I can't do the calibration process. In fact, the big lock knob has to be exactly at 12 to generate a new sequence. I've read that some people couldn't lock the sequence, but for me it's almost the contrary. The "randomizing" range on this pot is so short... when I try to do the calibration process, I follow the instructions but the output is directly stable and adjusting the trimmer has no effect, it only has effect when the "lock pot" is set to midnight almost exactly (I've checked the trimmer with a multimeter and it seems to be OK). I tried to re-solder some points and now the pulse output is not working :( that was my first DIY module. Last time I resoldered some points, I forgot to remove some IC. Do you think they are messed up now ? I have no prior experience and I'm not sure if they can suffer from the heat of the iron...

Elberstein, 2017-05-29:
The board has not been re-cleaned yet but here's a picture https://www.dropbox.com/s/85uhkzsptcp0720/Fichier%2029-05-2017%2021%2055%2020.jpeg?dl=0

Elberstein, 2017-05-29:
https://www.dropbox.com/s/7utc48emeirts3k/Fichier%2029-05-2017%2022%2000%2045.jpeg?dl=0

TomWhitwell, 2017-06-21:
Sorry to hear that - have you made any progress since this post? Does your noise output work properly? 
It's fine to resolder with the ICs in place, I do that all the time!

Elberstein, 2017-06-21:
The noise output is still not functioning and the randomizing range is still very short. I will try to reflow, now that I'm a bit more experimented with soldering. Or change the jack maybe ?

TomWhitwell, 2017-06-21:
If the noise circuit isn't working, that explains the problem with randomisation. 
This is the noise circuit, so have a look around the TL074 and the transistor on the front (pot) board. VR1 is the trimmer pot 
[image]

Elberstein, 2017-06-26:
Hi Tom, 
I reflowed everything and I replaced the TL074 with a new one. Now it's working. The noise output is working and the issue with lock pot is also solved. Thanks !

closed 2017-06-26 by Elberstein