TomWhitwell/TuringMachine #60: No Clock LED, No Output, build not working

bertog, 2017-05-31:
Hi,

I've just finished my second TM build and this time I've some issue. The module power up but when I plug the square signal in the clock input there's no LED activity or audio output (seems something but very very very low). The DAC0800 become also immediately very hot. Some advice to begin some trouble shooting?

Some picture of my build:

[image]
[image]
[image]
[image]

Thank you very much. Guido Bertolino

Thonk-Steve, 2017-05-31:
Do you still have your successful build?
Have you checked with a multimeter for a short between +12v and gnd, or -12v and gnd?

bertog, 2017-05-31:
Hi Steve,

Yes I've access to the other build. 

I've checked now and I've no short.

G.

Thonk-Steve, 2017-05-31:
I would suggest trying a combination of front and rear PCBs with the working build, then at least you can find out which board is faulty? You tested for shorts with the defective PCBs mated together?

bertog, 2017-05-31:
Thanks for the help.

I've matched the rear PCB of successful build with the jack pcb of the faulty and everything is working. I've successfully calibrated the module. 

So the problem is on the rear PCB. 

I've check for short with the mated PCBs and now only on the rear, but seems there's no shorts

G.

bertog, 2017-05-31:
Ok Steve,

I've founded the problem ... there was a small piece of lead touch one of the IC solder joint ... 

I've removed it and now is working ... 

Sorry and thank you very much for the help ... I'm newbie so I fall in panic very early, but tonight I've learned some troubleshooting ... Thanks again :) Guido

Thonk-Steve, 2017-05-31:
great, just talking about and trying new angles often shakes these things out :)

closed 2017-06-21 by TomWhitwell