TomWhitwell/TuringMachine #61: Calibration issue, no sound/or weird behavior

lemqmq, 2017-06-02:
Hi! I'm having strange issues with the turing machine. First off, if worked for more than 6 months without any sign of strange behavior. Then it started acting strange, having bits disappear until they're totally gone and nothing goes out. I'm now trying to recalibrate it but i noticed i get a steady tone only when the trimmer is fully CCW. Sometimes i don't get any sound out during the calibration procedure.
I tried a couple of things : swapping TL072, 4015 and the transistor. I also inspected every joint and reflowed everything just to be sure. I have no idea what to try next, really confused :( .

TomWhitwell, 2017-06-21:
Hi, sorry to hear that. 
- 'sometimes I get no sound during calibration' is normal - just let it randomise again
- calibrate it so you get a *nearly* steady tone, then test it in practical use as a sequencer

llumen, 2018-07-31:
sorry if this seems like hijacking the tread, but I figured keeping everything calibration related at one thread.
 @TomWhitwell, the calibration video says to turn the potmeter CCW until a steady tone but not completely CCW. With my newly build Turing machine, the steady tone only appears when the potmeter is turned completely CCW, can this be a normal behaviour, I'm feeding it with a square wave OSC at approximately C4-value. Thanks for the great module!

efluon, 2018-11-10:
Hi!

I have a similar behaviour, need to turn the trimmer almost fully CCW before a steady tone appears. 
This leads to a very narrow random range, making it almost impossible to dial in gradual changes.

I have no idea yet what to tweak. I did not build from the kit though, as i got the PCBs with the pretty micro hybrid panel from greyscale. Maybe there have been changes to the PCB or the BOM?
(Btw, these are very nice PCBs, fabulous layouting job there!)

Any hints?

Cheers,
Eike

> @TomWhitwell, the calibration video says to turn the potmeter CCW until a steady tone but not completely CCW. With my newly build Turing machine, the steady tone only appears when the potmeter is [...]