TomWhitwell/TuringMachine #63: New build not working (no bits being generated?)

digable-me, 2017-06-25:
Hi,

This Turing Machine MkII is my first build (after some practice bits and pieces) so I'm sure these issues are down to my amateur-ness.

So, this is the issue:

- When powered on with nothing plugged in the whole top row of LEDs lights up except the left most one.

- Clock input, pulse output, noise output, all seem to work fine.

- When I supply a clock input the top LEDs turn off in sequence but no new bits are being generated and the LEDs remain off until I push up on the WRITE switch. Then bits are generated on the LEDs and on the output, but no bits are subsequently spontaneously generated.

- With an audio rate CV input, listening to the main OUT, pushing the WRITE switch upwards creates a brief tone at the output followed immediately by a static voltage offset. The offset falls to zero once the WRITE switch is released. The LEDs remain lit for as long as the WRITE switch is held up, but the tone on the OUT is brief.

Here are some pics of my build:

[image]
[image]
[image]
[image]

Many thanks for any help. Really excited about this module and truly hope I can get it working asap.

TomWhitwell, 2017-06-25:
It looks like you've used a 40106 instead of a 4016 

The 40106 is a [super useful chip](http://milkcrate.com.au/_other/sea-moss/), but not in this module. 

[image]

digable-me, 2017-06-25:
Hi. That's all the module came with. I bought a full kit from Thonk.

TomWhitwell, 2017-06-25:
Oh, OK, sorry to hear that, I'm sure Thonk can get you a replacement - I'll also CC this thread to them. Good news is that I don't think the chip will damage the rest of the circuit, so you should just be able to swap in the right one.

closed 2017-06-25 by TomWhitwell

digable-me, 2017-06-25:
Ok thanks. I'll send them an email now.

digable-me, 2017-06-25:
And many thanks for the super quick response.

Thonk-Steve, 2017-06-25:
Hi digable-me, I thought i'd tracked all these down so sorry about this. drop us a mail on support@thonk.co.uk

The good news is the chip won't have damaged your module.

digable-me, 2017-06-25:
Thanks. Email sent.