TomWhitwell/TuringMachine #64: Clock LED doesn't trigger and all lights come on at the top, don't change.

calmoo, 2017-06-27:
So i finished the build today. I've double checked all components and solder joints.

When I turn it on, all the lights at the top of the Turing Machine come on. The clock LED doesn't respond to a clock source at all. Sometimes when I turn it on, the top left LED stays off while the others stay on. The knobs don't change anything. The noise output works.

Not sure what I'm missing?

TomWhitwell, 2017-06-27:
Check around the TL074 on the front board - that's the clock circuit - it's possibly a missed solder point there or on the clock input. If noise is working, then that TL074 is working - receiving power etc, so it's likely to be a missing solder joint or component. 
[image]

calmoo, 2017-06-27:
Figured out that the clock source I was feeding from my phones LFO
generator volume  was too low. Daisy chained it from a doepfer envelope
generator and it worked a charm. Thanks Tom!

closed 2017-08-03 by TomWhitwell