TomWhitwell/TuringMachine #67: can't calibrate, get a kind of sinus sound...

Latzenpratz, 2017-07-05:
Hi!

just finished my build. I'm trying to calibrate, but the main pot and the calibration pot don't do anything. Length, write and scale seem to work. I here something like a sine wave played with  the set length, but no randomness.

Thonk-Steve, 2017-07-06:
Would it be possible to upload a video demonstrating your problem?

Latzenpratz, 2017-07-06:
Thanks for the quick response!
I'll try to make a video next weekend...

Latzenpratz, 2017-07-13:
Hi! 
I noticed that the Turing triggers my maths without connecting it (i can hear the turing output from the maths output). The only connection between them is the power board...
So i'm to scared for now to reinstall the turing and make a video.
Maybe you can see an error on the pics:
[image]
[image]
[image]
[image]

Latzenpratz, 2017-07-21:
On my MDLR busboard the -12v LED doesn't light up if i install the turingmachine, so it seems there is a short-circuit or something... help...

Thonk-Steve, 2017-07-21:
Use a multimeter continuity tester to see if you have continuity between:

+12v and gnd
+12v and -12v
-12v and gnd

Latzenpratz, 2017-07-21:
yes, continuity between -12v and gnd

Thonk-Steve, 2017-07-21:
yeah I would start looking along the -12v lines and see if you have any
shorts

Latzenpratz, 2017-07-21:
yes the short must be on the board with the music thing print. i'll do my very best ;-) THANKS!

Thonk-Steve, 2017-07-21:
yeah it is worth taking the PCBs apart just to double check which board
it's on.

Latzenpratz, 2017-07-21:
I can't see any visible short, where are good points to break the circuit to narrow down the place of the short?

Latzenpratz, 2017-07-25:
Found it!  :yay: 
One diode was faulty, after removing it the turing machine works!

closed 2017-07-25 by Latzenpratz