TomWhitwell/TuringMachine #68: No Random output

Paulhamlin, 2017-07-05:
Hi Tom / Thonk

Thanks for putting together a great build with instructions, its been a breeze up untill now. I've come to the end of my build and all appears to be working well and behaving as should apart from the random voltage out. The scale knob adjusts pitch correctly and the lights displaying the cycling of notes however the outputted pitch is always the same. 

Have you any ideas about this or where would be a good place to start looking? Ive double checked solder joints and will triple check in the morn just to be sure. 

Thanks 
Paul

Thonk-Steve, 2017-07-05:
Hi Paul, is the white noise output working?

Paulhamlin, 2017-07-06:
Hi Steve, thanks for getting back to me - yes im getting white noise when connected directly to an output.

Thonk-Steve, 2017-07-06:
Hi Paul,

Is the white noise output a relatively 'normal' volume level? not super
quiet or anything?

You're sending the main output to the 1v/oct input of an oscillator to
test? what are you clocking it with?

Steve

Paulhamlin, 2017-07-06:
Yeah the white noise level seems good. 

I'm clocking with maths that is going into the 1v/oct on braids. Just metered the output with the scale knob on zero and the module cycling and the output shows a steady 0.00v - moving the scale knob moves this voltage up however there is no fluctuation in voltages. 

Cheers

Paulhamlin, 2017-07-06:
Spent a few hours earlier today checking over the board / checking for shorts and making sure that i didn't put anything in incorrectly and I'm still getting the same behaviour.

From looking over the schematic it looks like the DAC chip is responsible for the output, could it be possible that its faulty? Ive searched high and low on the Muffwiggler forums and can't seem to find anyone else with this problem...

Any help or advice would be much appreciated! 

Thanks

TomWhitwell, 2017-07-13:
Sounds like a problem with either the DAC or the TL074. If you have the clock and noise working, they're both on the TL074, so it points to the DAC, or the surrounding infrastructure: the TL431 voltage reference, R28, R27, C9, R26. 
Completely unchanging output means it's not getting partial data, so the 8 data lines into the DAC can't be the fault. Could be a bad DAC chip, or a soldering error  / short somewhere around there.

Paulhamlin, 2017-07-19:
Hi Tom, 

Thanks for the advice - managed to solve it after metering between the parts you listed. 

Turned out a pad had somehow got damaged and wasn't flowing to the PCB solder - putting a cable link in got it up and running, not the prettiest but got it working correctly!

Thanks again
Paul

closed 2017-07-19 by Paulhamlin

guidozzz, 2019-01-20:
Thank you Tom (and Paul for starting the thread with exactly the same issue as me) !
A connection between R27 and TL431 wasn't working. I have done a wire link and seems to work fine now!