TomWhitwell/TuringMachine #69: No Noise Output

jamesbvaughan, 2017-07-07:
I just finished assembling my module and I'm not able to get any noise out of it. Most of the functionality seems to be working properly, except for the noise and the randomness (but I'm guessing that the lack of randomness is being caused by a problem with the noise.)

I can't find any obvious problems on the boards. What tips do you have on debugging this problem?

TomWhitwell, 2017-08-03:
The issue with noise will be somewhere here - no output at all means it's likely a missed connection rather than a strange value somewhere: 
[image]

DaJensemann, 2018-03-13:
I am having the same issue and I am getting frustrated, so I am having problems being objective. I exchanged the 2N3904 and the TL074 but no change. Reflowed all the connections. 
Can someone tell me, what the noise level from the transistor should be? That would allow me to set up  small generator to trace the signal and see what is wrong with the signal path. (After this I am going back to SMD, I am getting to old for through hole :)  
Thanks in advance.

Thonk-Steve, 2018-03-13:
Do you have an oscilloscope? Are you seeing noise at any point in that part
of the schematic?

DaJensemann, 2018-03-14:
Thanks for replying so fast Steve,
Yes I checked it with the OSC and was not able to find get any signal from it that's why I asked for the levels. It looks more like the signal is connected to ground, but I cant find the bridge if there is any. 
Do you think it make sense to remove all the parts involved and start this part from the beginning?

nnay13, 2018-03-19:
Exactly the  same  issue here . I have reflowed all the connections.
I have checked all the resistors and the caps in place for what it's worth ...  I haven't checked  with an oscilloscope yet . I have to find  one.  
Any advise  on the troubleshooting ?

DaJensemann, 2018-03-19:
I had no time to desolder the parts and see what exactly is wrong with it but I will report back. How hard can it be?!
:D

nnay13, 2018-03-19:
I just tried continuity testing  with a multimeter .  I  found C18 and R31  to be connected to ground. With or without the TL074 in the socket.  I found no obvious solder  bridge. 

Should I unsolder C18 and R31  and resolder(replace) them ?  

Any help would be appreciate.

DaJensemann, 2018-03-19:
I doubt that the resistor is your problem. 
Is pin 9 of the TL074 Socket on GND too?

Thonk-Steve, 2018-03-19:
DaJensemann and Yann, what PCB version numbers do you have?

DaJensemann, 2018-03-19:
Rev 4 for the Front and Rev 4.1 for the Backboard

nnay13, 2018-03-19:
The same : Rev 4 for the  front   and  Rev 4.1 for the  Backboard  .

Thonk-Steve, 2018-03-19:
ok don't think it's a specific PCB issue then, we have tons of confirmed builds with those. Uploading pictures of both sides of both PCBs can help us spot something you've missed.

nnay13, 2018-03-19:
Pin 9 isn't  grounded.

nnay13, 2018-03-19:
I don't know if the picture are good enough .  

[image]
[image]
[image]
[image]

DaJensemann, 2018-03-19:
Thanks Steve, that is good too know.  And here are my photos:
[image]
[image]

nnay13, 2018-03-20:
Peharps these  are better :  

[image]
[image]

nnay13, 2018-03-20:
I've just  checked continuity   without  the TL074  in the socket . PIN 8 goes to ground  as C18 and R31.   What should  I check on the  PCB ?  Should  I remove the IC socket  C18 et R31

TomWhitwell, 2018-03-30:
I’m not sure I understand. Pin 8 should connect to c18 and r31. It should not connect to ground (ie the centre 6 pins on the power header)

jstamant, 2018-04-01:
Wow, it's a little tough troubleshooting two issues on a hijacked thread. None of these issues are necessarily related. I can see the issues are definitely similar, but not necessarily identical. No matter, though, here's some troubleshooting ideas:

@nnay13. You have to find out why pin 8 of the TL074 socket, C18, and R31 are grounded. No need to find an oscilloscope while you've got other, much more prominent issues, such as continuity. I checked the board layout file, and I don't see any nearby connections to ground that could accidentally be solder-bridged. Also looking at the pics you sent of your soldering, looks like the soldering has no bridges, as you've reported. My suggestion is to desolder and remove each component (TL074 socket, C18, R31), one at a time, and check continuity from that net to ground each time a component is removed. Another thing you can do is check the continuity of everything in the noise generator circuit. Make sure that each component's pin has continuity to every other pin that it's supposed to connect to. Earlier in this thread you mentioned that you "checked all caps and resistors in place, for what it's worth." Can you please elaborate on what that means?

@DaJensemann. To confirm: you are not getting any noise on the noise output jack? Was your Turing Machine working correctly at one point, and this is just a latent failure? If this is a new Turing Machine, your issue is likely a connection somewhere, so there shouldn't be a need to take out the ol' function generator and scope. As Tom mentioned months ago to the original poster of this issue, the problem is likely with the noise generator circuit. You should check continuity between all the connections in the noise generator circuit if you haven't done so, yet.

nnay13, 2018-04-04:
Sorry for  the late  reply. Many thanks for your help. 

@TomWhitwell : I  did check continuity  between each node in the noise circuit and a leg of the  50k pot and/or the  outer  ring of a jack not the center  6 pin of the power  connector .  

@jstamant : I have unsoldered  C18  and destroyed it . I'm waiting for a remplacement.  :(
I checked the caps values and the resistors  values  with a multimeter  after  soldering  them  on the PCB not before soldering .  I'm not  fluent  in english. That's what I meant by "in place".