TomWhitwell/TuringMachine #70: Led B4 not working

Rolf102, 2017-07-26:
Hi,

I finished my Turing machine mk2 today it is working perfectly except for one led, i checked the schematic, it is led B4.
I tested the LED it is not broken, i reflowed all the joints on both boards with no effect.
While it is a minor defect the perfectionist in me cannot overcome this.

Please help

TomWhitwell, 2017-07-26:
Did you check the LED is not reversed?

Rolf102, 2017-07-26:
unfortunately no all the leds are in the same orientation

Rolf102, 2017-07-27:
i checked all the components and found that i soldered one of the 78l09 with the flat face to the wrong side, i fixed it but the led still does not light up, should i replace the 78l09 for a new one?

Thonk-Steve, 2017-07-28:
do you have a multimeter? can you check if the new 78l09 is putting out 9V?

Rolf102, 2017-07-29:
both voltage regulators are putting out 9 volts

Rolf102, 2017-07-29:
i fixed the problem, the trace from the led negative pole to the resistor is broken.
i soldered a copper wire from the led to the resistor

closed 2017-08-03 by TomWhitwell