TomWhitwell/TuringMachine #71: TuringMachine not working at all

dorkajef, 2017-08-03:
Hello,
Finished My DIY for the Turing machine yesterday. But it did not show any traces of life -> no noise, it did not blink lights, did not effect the OSC when patching, did not react to clock signal.
I did check My solders couple of times, double checked also the parts -> everything should be ok (but after one check the upper light when on ((but no reaction to clock, etc...) -> and when I unplug the device and put all the nuts on -> again no lights.
I know that My solder is not the best A+, but all My DIY units have worked until now.
Would you have an idea what´s wrong?

Best Regards
Thomas
[image]
[image]
[image]
[image]

TomWhitwell, 2017-08-03:
Hi, sorry to hear that. When I have problems like that, it's often because I've done something extra stupid - like not turning on the power, not connecting the other end of the power cable etc. 
One thing you could try is - if you have  multimeter - checking that you're getting the appropriate power to the chips - i.e. separate the boards, plug the back board into power, and check there is (roughly) 24v difference between pin4 (black probe) and pin8 (red probe) on the little TL072. That will tell you if there's a basic problem with power. If the back board is OK, carefully check the soldering on the connector pins between the back and front boards - if the front board isn't getting any power, you'd get the symptoms you describe.

dorkajef, 2017-08-03:
Thanks Tom for the Quick answer. 
Gonna try that tonight after I come home. 
Best
Thomas

dorkajef, 2017-08-03:
Hello,
I think I may found the problem. There is a shortcut in the six pin header (the middle one on the back board). Could this mistake have fried my parts?
Thanks!
-Thomas

TomWhitwell, 2017-08-03:
Possibly not - that connects +12 to -12v, which I'd imagine would cause your case power supply to shut down quite fast - fix it, check there's no more connectivity, and test to see

dorkajef, 2017-08-03:
Hi,
Just to make sure that we are speaking from the same part... 
I removed the connection part, and try to clean the area, but if I assume right the PCB has jumpers between pins 1&4, 3&6, and on the other side 2&5 (but I quite n00b with these things;).
I´m just wondering because my meter is beeping with these connections (also with the upper PCB).
Best Regards
Thomas
[image]
[image]

TomWhitwell, 2017-08-03:
This is how they're connected: 
[image]

dorkajef, 2017-08-03:
Sorry my bad, I did mistakenly assume who those are counted -> so no shortcut there, they where as they should be. 
I did also measure the power as You instructed -> and I got a reading of about 25 Vdc and 52 Vac.
Did also solder all the connector pins newly (at that moment I mistakenly noticed that so called shortcut). So any ideas?
Best Regards
-Thomas

dorkajef, 2017-08-03:
Hi,
I did also test the connection pins while the boards are connected, the connections works - but I also noticed that on the upper board has also an connection beteeen +12 & GND somewhere (this is normal, I suppose).

-Thomas

TomWhitwell, 2017-08-03:
No, there shouldn't be a connection between +12 and GND - that would also the problem.

dorkajef, 2017-08-04:
Ok, I´m on the trace. 
I did try to find out where the GND goes (but can´t see it from the PCB), and I´m not enough seasoned to read it from the schematics. So if you can advice Me (then I maybe could find the shortcut).
Thanks
-Thomas

TomWhitwell, 2017-08-04:
GND lives on the ground plane - it's all the copper on the bottom that isn't part of another trace: 
[image]

dorkajef, 2017-08-04:
Ok, thanks.
So non of the GND points should make a connection with +12?
-Thomas

TomWhitwell, 2017-08-04:
Definitely not

dorkajef, 2017-08-04:
Hi,
Have been looking for any solder shortcuts (could not find any). I even removed some parts (all capacitors but not: c17, c14 & c9).  The IC´s I removed in the beginnin of this operation. 
Still a shortcut when measuring Link2 pins 6&4 (+12 & GND).
Any idea where / what path I could try to find next?
Thanks for the help.
-Thomas

TomWhitwell, 2017-08-04:
Do you think the back board (with the power connector) is OK? Is that's where you tested the PCB voltages on the TL072?

Separate the boards, remove the two ICs from the front board, and see if there is still a short between +12v and GND

dorkajef, 2017-08-04:
Yes that is what I did. I assume that the back board was ok (measured the voltages as described -> ok), the pins had no short between the pins (when measured the back board separate)- > So I´m quite sure the problem is on the front panel.

closed 2017-12-19 by TomWhitwell