TomWhitwell/TuringMachine #73: Input clock works correctly for 15-20 minutes, then stops receiving signals

MatteoGgl, 2017-09-20:
I recently bought and built a TM mkII myself from Thonk. When I plug the module in and clock it with a square bipolar wave from my DIY Peaks everything works, but after 20~ minutes of fun the module starts behaving incorrectly: initially the clock becomes irregular and starts advancing many steps with only a single pulse and after a minute or so it just stops recognizing any kind of signal in the clock input and just stops working. I changed the clock jack hoping the one I soldered first was defective and power-cycling the rack gives no solution; also inputting a different wave (even from a VCO, Braids) does nothing. It almost seems it has to "cool down" before it can be used again! What can be done to fix this amazing piece of equipment?

TomWhitwell, 2017-12-19:
That is very mysterious - did you ever find a solution? Sorry I've only just seen this today.

MatteoGgl, 2018-02-23:
Yeah, I know, I was baffled for a while. Now it's working normally!

closed 2024-12-17 by MatteoGgl