TomWhitwell/TuringMachine #77: No sound out

chris-roerig, 2017-10-06:
Everything appears to be working except for sound out. I'm getting noise from the noise output and clock etc all seem to be working. What part of the circuit manages the sound out? I'm not sure if its a bad IC or what. Thank you!

Thonk-Steve, 2017-10-06:
Hi Chris, the Turing generates CV, not sound, what are you patching the
output to?

chris-roerig, 2017-10-06:
Tom, thanks for the reply. After removing my palm from my face, I hooked up the "out" to the "vco 1v/oct" on my Mother 32. I've also run a gate to the TM. I have the VCA in full "on" mode but the TM is not altering pitch at all. What could be wrong here? I've replaced every IC and looked over every join on this thing. That doesn't mean its perfect but I'm running out of places to look. Thank you for your help!

eeweegh, 2017-10-06:
possibly silly comment, but the scale knob is not fully ccw? I can't
remember whether that fully turns off variation on the pitch output (and
can't test it at the moment).

chris-roerig, 2017-10-06:
@eeweegh, thanks for the comment. I double checked the scale knob with no success. Seems with all my poking I broke the led that flashes along with gate input :/. the LED doesn't look burned..

closed 2017-10-11 by chris-roerig