TomWhitwell/TuringMachine #79: CALIBRATION problem

BORSCH2503, 2017-10-31:
Hi!
Just finishing my module! Everything works fine. Pulse out works.I hear a noise from the noise output between pulses.Is it ok?
But i have some problems with the calibration.
When i try to calibrate square tone became stable only when trimmer pot is completely CCW.
And something wrong with lock function-it locks on 12 o clock and make random on CW position.

TomWhitwell, 2017-10-31:
"And something wrong with lock function-it locks on 12 o clock and make random on CW position"
This sounds like some kind of construction error - a missed solder point, or a short circuit somewhere.

BORSCH2503, 2017-11-01:
When i listen to noise output and rotate the trimmer noise volume don't decrease as i think it should!
Its go to silent only at full CCW position and full volume at any other?
Works like a on/off!
Any thoughts?

TomWhitwell, 2017-11-01:
Again, check the soldering - look for any short circuits or chips the wrong way around.

adiogo08, 2017-11-16:
@TomWhitwell hi Tom.  Just finished my build as well.  I can't seem to get it to calibrate with out being all the was counter clock wise.  Could this be to not using a proper VCO square wave.   I have a erica synths wavetable and a make noise sto.

TomWhitwell, 2017-12-19:
@adiogo08 See if the module works properly as a sequencer at normal LFO rates (rather than audio rates for calibration) - it should clock OK to the STO or Erica Wavetable.

closed 2017-12-19 by TomWhitwell