TomWhitwell/TuringMachine #80: Module partially not working

Stahlmotte, 2017-11-19:
Hi, 
first of all: I´m not an expert in DIY modules. :-) So it would be great to get a little help. Maybe this is also something that happend in the past and a quick fix.

At the moment noise out works and all the LEDs (besides the first one) in the top row are on permanent. Nothing else is happening.

I checked for solder joints and stuff like that. I also checked the placement of the items. No luck so far. 

So a little help would be great!!

Thanks in advance

Alex

TomWhitwell, 2017-11-19:
Hi Alex, sorry to hear that. 
What are you using to clock the module?

Stahlmotte, 2017-11-19:
Hi Tom,
thank you for the quick reply. I tried a the Black VC Clock Module von Erica Synth and also Clock signal via uMidi. Anyway, sending a clock signal does not change the behaviour anyway. With or without clock signal.

TomWhitwell, 2017-11-19:
So your issue might be around the clock input - if the clock in doesn't flash the LED above the clock socket, that is definitely the place to start looking - check for missing solder joints or IC in the wrong way around. 
I've often built a whole module, found a weird fault, then discovered I missed a couple of solder joints on sockets or ICs. 

Schematics are here: https://github.com/TomWhitwell/TuringMachine/blob/master/Collateral/PDF_Schematics%20/TuringMachine2_Schematic_May2016.pdf 
If the noise output is working, then the TL074 is alive (i.e. connected to power and the right way around) so it could be a connection to it, or one of the other components.

Stahlmotte, 2017-11-19:
Hi Tom,
tried to find all faulty solder joints in the last days and also today. Sure, i might be missing something. :-) Anyway, do you think it could be also a faulty IC? Thinking of ordering and swapping them if this could also be a possible option. 
Best
Alex

TomWhitwell, 2017-11-19:
If you don't get a flashing LED above the clock in when you input a clock, that along will mean the rest of the module won't work - everything else could be perfect. The only IC involved in making that LED flash is the TL074 on the front board, and if you have noise, that IC is at least partly working. 
In short - I think there's a faulty connection or a misplaced component somewhere here: 
[image]

Stahlmotte, 2017-11-20:
Tom,
thank you again for your help. I really appreciate that! Yesterday i resoldered several joints on the board. Now, when powering up the module the LED flashes once at least. Everything else is still the same, but i will try to wirk through all the joints again. Hopefully this is the solution!
Best
Alex

closed 2017-12-19 by TomWhitwell