TomWhitwell/TuringMachine #84: Possible Issue / Not Sure

DJmerkury, 2017-12-23:
First off, thanks for your work. My Turing Machine, Radio Music, and expanders are awesome.

I am not sure if this is an issue or just normal behavior. Quite often, when I have been looping a sequence, I rotate the main knob back to 12 o'clock and I end up getting either a stream of all high or all low. In other words, the led lights illuminate one at a time until they are all lit or the opposite. The pitch coming out also goes to a steady monotone high or low signal. If I leave the knob at 12 o'clock for a a few seconds, it will "recover", but I am now questioning if this is normal or not. If it is not normal, what should I look for or replace?

Thanks!!

jstamant, 2018-03-22:
> I rotate the main knob back to 12 o'clock and I end up getting either a stream of all high or all low.

> If I leave the knob at 12 o'clock for a a few seconds, it will "recover", but I am now questioning if this is normal or not.

@DJmerkury From the sounds of this, there's something wrong with the noise circuit, or the knob+quad-bilateral-switch circuit. When you rotate the knob to 12 o'clock, the TM should immediately start clocking in random bits. It shouldn't clock in a stream of 1's or 0's before "recovering" to it's expected state of clocking in random bits. That's what I understand from your description of your TM's behaviour, which is not normal.

When you check the noise output of the module, does it sound like white noise, or are there any glitches? Have you performed the calibration as refered to in the build document, and was it successful? When using the write switch, does it insert a bit or remove a bit successfully on each clock pulse? Did you get a kit from Thonk? Did you place all the capacitors in their correct location? Did you place all resistors in their correct location? Did you encounter any other issues while building this module, or did you require any troubleshooting to get the module working?

Many questions, but I have to know what you've already done.

DJmerkury, 2018-03-23:
@jstamant Thanks for the reply. To answer your questions
White noise sounds good. Looking at it on a vst spectrum analyzer, it has less gain at lower and higher frequencies (sort of a rounded off look with lowest frequencies starting at around 30 Hz. Peak volume at 1000 Hz and about half of peak at 16k Hz. It looks and sounds pretty good honestly. I see and hear no issues with white noise.
I did perform the calibration with absolutely zero issues.
The write switch does work 100% correctly.
I bought the kit from Modular Addict in the USA. It had a thonk sticker inside the bag, so I would imagine that it made it's way here from thonk.
As far as I know, I built it all correctly (resistors and caps all where they belong). I am pretty meticulous when building a module (checking off each component as I go along).
No other issues and no troubleshooting to make it work.

After answering all the questions and thinking more about it, I think it is probably working correctly. I have read a little more about it since posting here and understand a little more about how it works.

Thanks again for answering and feel free to comment further on the issue if you think it may truly be of help.