TomWhitwell/TuringMachine #85: Powers on but not much else

schrickvr6, 2018-01-23:
Hi Tom,

I'm a relative noob to DIY and modular so please bear with me, I bought panel and PCB so there's every chance I've bought an incorrect part.

Basically it powers on ok, it may be inconsequential but sometimes all of the top row of LED's light up and sometimes a few don't. I'm getting no output (except a tiny bit of noise when I adjust scale), no movement or randomness but most tellingly it seems no input on clock, I've tried a square wave as in the calibration video and a gate from a beatstep with no response or clock LED activity.

In the way of fault finding I've reflowed everything, reseated IC's and swapped matching pairs, I've also measured all the resistances which tally with marked values except R10 which measured 83k instead of 100k, but various people on Facebook have reassured me that should be normal.

Any thoughts on where my problem may lie or how I should start/continue fault finding?

Cheers.

schrickvr6, 2018-01-25:
Please ignore (for now), I'd fitted a 68ohm instead of 68k right next door to r10, I don't have a spare to confirm this is my only issue but will close this if it works when my resistors arrive.

TomWhitwell, 2018-01-25:
Sorry to hear this, Yes, R10 as 68R vs 68K that would do it - I was going to suggest you look at the TL074 on the front board, but if you're pulling the clock to GND you won't get much joy.

closed 2018-01-25 by TomWhitwell