TomWhitwell/TuringMachine #87: Random - but not really

doktorfilter, 2018-02-16:
Hello, 
I need help with my Turing Machine mkII. I´ve basically read every single thread in this forum but nobody seems to have had the same issue.

The problem is that I can barely get any usable pattern out of the machine. There is not a  wide range of notes generated as I´ve seen in several demo videos. Evry time I clear the buffer ashort descending pattern is generated and leads again to the very narrow range of notes after that. The scale poti doesn´t transpose to anything usable after about 3 o´clock even if I tune the OSC to the highest frequency. Also calibration was only possible in the 100% CCW position of the calibration screw.

As you can see the issue is pretty hard to explain as every component of the device seems to basically work. The random generation of CVs behaves a litte strange and unexpected though (which seems paradox to complain about, I know). To illustrate the issue better I recorded a short video to demonstrate the kind of behaviour I´m talking about: https://youtu.be/Jl3tebSIu-A

I´ve already checked every single solder point, even resoldered some of borderline-perfect ones and of course checked the orientation of the ICs several times. I could tkae some photos if that would help aswell...

As you can also see in the video my build of the Pulses expander doesn´t generate pulses through the first five outputs. I´m not looking for help with that issue in this thread but maybe this might be another hint regarding the issue with the random behaviour.

Thanks a lot in advance!
Malte

DJmerkury, 2018-02-20:
Just curious, what sort of sound are you getting out of the noise output? Is it typical white noise? I watched your video and yours does not do what mine does. I will tell you that I had issues with my pulses module. Seems like I had two outputs that were not working and when I tracked down the issue, it was solder joints on resistors or the LEDs near the outputs. I posted a thread about 2 or 3 months ago and never got a reply. Tom seems like a great guy from everything I have seen or read about him, but the response on github is not too good. I hope you figure it out. Get the schematic and start checking over everything you know how to check. You'll find it eventually.

doktorfilter, 2018-05-11:
Hey,
I´m very sorry for the late response but I had some things to deal with away from the modular world :)
So, the noise seems fine. I didn´t measure it but to me it´s plain white noise.
Thanks a lot for the advice! I don´t have an oscilloscope but if I hab one I wouldn´t really know what I´m looking for when checking pins. I have also checked all solder joints with focussing on the parts in the schematic that seem to be more relevant to the issue but as it´s such a weird behavior I´m not really sure which area might be best to check more in detail. Does anyone have any idea which approach would be the more obvious?

TomWhitwell, 2018-05-11:
From the video it looks completely normal in terms of locking / random, BUT there is something very odd with your output - does turning the scale pot clockwise create *lower* notes? That is very odd behaviour.

doktorfilter, 2018-05-11:
Hey Tom,
thanks a lot for the quick reply!
Yes, that is the case. That´s actually what I meant by '__The scale poti doesn´t transpose to anything usable after about 3 o´clock even if I tune the OSC to the highest frequency._'

TomWhitwell, 2018-05-11:
OK, that's completely wrong - 
Schematic is here: http://musicthing.co.uk/collateral/TuringMachine_schematic.pdf
If you look top right on the top schematic, that's the CV output section- the DAC chip, and the TL074 which drives the output. 
Places to check: 
- Check all the pins on the TL074 chip on the front board - they're soldered OK, the chip is seated in the socket OK, there are no bridges. Something like missing the GND going into pin3, or a missing connection on R26 might do this. 
- Check connections on the output pot 

Basically everything here: 
[image]

doktorfilter, 2018-05-11:
Okay thank you for the direction, so I just resoldered TL074, The DAC and R26. At some points the solder hadn´t gone through perfectly before. Nothing changed though after the adjustment. I´m not sure what you mean by output pot, I´m sorry. What exactly should I check?
Just to be sure, the normal behaviour of the scale poti is expanding the key range of the random notes, right?

Thank you for your time by the way! I really didn´t want to bother you until I had no clue what to do anymore. Much appreciated! :)

doktorfilter, 2018-05-12:
So, I checked and partially resoldered every single solder point again. To my eye everything looks perfect now :) (I attached some pictures of the current state.)

No change of behaviour though. The scale pot still pitches the notes down and only very few different voltages are generated. When turning the pot all the way to 5 o´clock, my multimeter says that about 90% of the voltages are around 8V and the fews changes cover about the whole spectrum to almost 0V. The generated pitches are so low that they are at the bottom of the audible range with the VCO (Mutable Braids, tried other VCOs aswell) tuned to the highest possible frequency which I don´t understand. Shouldn´t higher voltages translate to higher pitches? And do you think the distribution and the frequency of changes are normal? I´m so confused right now.

Here is a recording of the scale pot at 5. At the End of the recording I´m turning the pot successively all the way to 7: [https://soundcloud.com/malte-seddig/turing-scale/s-tMnWw](url)
Can you make any sense of this?

[image]
[image]
[image]
[image]

kimitobo, 2019-04-13:
Chipping in here. I had exactly same issue (Scale pot effect reversed and poor scale in randomness) and the problem turned out to be bad solder on R26 (leading to Pin2 on TO74). So excellent troubleshooting above. Thank you! One problem remains, sharp spike on almost all CV changes, posting another issue on that one.

doktorfilter, 2019-04-20:
Thank you very much for your input! Finally someone with the same issue. :) 
I´ll have another look into R26 and hopefully close this issue then...

kimitobo, 2019-04-20:
Hope you find it. I had couple of resistor values swapped and had to resolder R26. The trace to the pad got damaged a bit after wicking it so I scratched the trace a bit to expose the metal and resoldered.