TomWhitwell/TuringMachine #89: Shift register dropping bits

jstamant, 2018-03-21:
I finished building my Turing Machine, and it mostly works, but now I've been troubleshooting some odd behaviour. Here's a weird one for you guys:

# Issue

The Turing Machine introduces and drops bits in the middle of a sequence, regardless of being locked or unlocked. The Turing Machine favors dropping over inserting bits, so a locked sequence quickly turns into all 0's.

Possibly related to issue #37.

# Troubleshooting steps taken

- The TM calibrates properly when clocked with a VCO. The glitchiness settles, and the TM is left outputting a solid tone.
- When the TM is locked, it works properly by not introducing any randomness on the first LED.
- When using the TM as a sequencer, that's when the bits start dissapearing between the 5th-8th LED.
- The dropping/adding of bits isn't completely random. If I send a single bit, it progresses all the way to LED 5, then dissapears on the following clock edge (nothing on LED 6). Always. 100% of the time. If I send different patterns using the write switch, the bits drop/appear differently, but only between the 5th-8th LED. It's like watching Conway's Game of Life in a straight line. A given bit will appear or dissapear depending on the presence/abscense of the surrounding bits.
- I've tried swapping the two 4015 shift register ICs. Same problem occurs.
- I've tried using one shift register at a time. Same problem occurs with both 4015 ICs provided.
- I've tried removing the 4050 buffer ICs and 4081 AND gates to further isolate the issue. Same problem occurs.
- Removing the 4016 switch IC also does not clear the issue.
- I've tried swapping the TL072 with a known working replacement. Didn't fix the issue, but that was expected, as the lock pot works fine.
- Swapping the TL074 with one from my spring reverb kit did not clear the issue, either. The noise output jack seems to work fine, outputting white noise.
- I've double-checked everything on my build. I've confirmed the placement of all the components against the board layout file, as well as confirmed their orientation.
- LEDs are all working.
- Diodes are pointing in the same direction as the silkscreen, and matching the board layout file.
- No 1N4001 blown.
- 9V regulators are putting out 9V.
- +/-12V rails are reading +/-12V with ~20mVAC.
- When running the TM in 2 or 3 step sequence length, you can see that the same 2 or 3 bits repeat, but the bits that trail past the first 2 or 3 LEDs go all wonky between the 5th-8th LED as previously mentioned. A 3 step sequence actually turns into a 6 step sequence with the extra adding/dropping in the shift register's 2nd unit.
- Powering up the TM some other time yields different results when writing a single bit. Before, a single bit would make it to LED 5, then dissapear. Now, a single bit makes it to LED 5, skips to LED 7, then dissapears.

# Further troubleshooting steps

There is still more I can do to troubleshoot this issue, but I wanted to get this lengthy post up to let people digest this while I take these remaining troubleshooting steps:

- Check continuity on everything. Starting with power rails, then the 4015 shift registers, then the DAC0800, then the then the rest.
- Scope the inputs and outputs of the main shift register for expected behaviour. Monitor the supply rails and reset pin for unusual bumps.

Last-resort steps:

- Unsolder resistors to measure resistance and/or replace them.
- Replace caps with equivalent caps that I have on-hand.

# Comments

As mentioned above, issue #37 seems related, but I'm confident it's not the noise section that's an issue. The shift register shouldn't be adding, dropping, skipping, or grouping bits together. It should only be shifting them down. If I'm not mistaken, noise/randomness only shows up on the first LED and shifts down.

I am experienced with soldering, and this build went very smoothly from the Thonk kit I purchased. The boards are good quality, and I did not damage the board or any traces throughout the build. I've checked the solder joints since someone will be asking for it, and they all look perfect.

Some background on myself: I'm a telecommunications technician during the day, so I troubleshoot things all the time; and then I'm an electronics engineer at night, so I have no problem releasing some magic smoke. I hope the above information is helpful for you guys.

QrchackOfficial, 2020-06-21:
I have the exact same issue. Initial testing shows the connections are there, noise output works fine. I've checked the TL072 connections, as well as 4016 ones

But what I found concerning is in this section:
[image]

I'm getting something more like +8.5V on R23, despite +12V_FRONT being +11.9V. Also something near -8V on -R35. Is this normal? During calibration the issue is, I get noise at 12 oclock, going either way results in a somewhat stable tone (or very stable, depends on trimmer position - but to get it fully stable I have to turn the trim almost all the way down, to the point of no noise coming from the noise output). However, after I try locking, get back to random and lock again, I always get exactly the same sequence. There is noise at 12 oclock, but when locked I get the same result every time.

QrchackOfficial, 2020-06-21:
For the record, all logic is CD4xxx, TI-branded. DAC0800LCN seems to be National. Sourced from China because no new stock seems to available, so I'm not very sure of them, but the shift register part should be working regardless of the DAC.

QrchackOfficial, 2020-06-21:
Testing this part of the circuit in LTspice shows there should be indeed +/- 8.5V on R23/R35, and there should be a range of +/-6.5V on LOOP_CTRL_FRONT. I'll test that and report back.

QrchackOfficial, 2020-06-22:
Interesting finding - I'm seeing constant 360mV on START_OF_LOOP (first shift register input) with 4016 removed, and constant 1.5V with the 4016. The write switch middle pin is behaving like a bistable switch - it keeps the high/low voltage even when released to center off position, and the only way to change it is by flicking the switch to the opposite position. The switch is fine, confirmed with a multimeter. R16/C4 are fine and connected to respective nodes in the circuit.

The shift register also loses bits after a couple of repeats.

QrchackOfficial, 2020-06-22:
Correction: this depends on the LOOP_CTRL position. With it fully to the right and the length set to 8, I'm able to get a single bit to loop - it skips the second LED every time and jumps straight from 1 to 3 on clock

QrchackOfficial, 2020-07-19:
I did some more testing. Tested voltages for the loop control:
[image]
And the comparator with noise. While calibrating, the point when I'm getting a tone that's somewhat stable (but drops out at times), the noise is about -5V to +5V. Voltage coming on LOOP_CTRL_BACK is +7.75, and the inverting input at the comparator is -3.65V, with the loop knob fully clockwise as in calibration instructions
[image]
Here's a scope datalog for NOISE_BACK (pin 5 of IC7B) on CH1 (yellow). This looks to be mostly -5V to +5V, with occasional peak up to 9V. There's also pin 7 of IC7B, which is the comparator's output on CH2 (blue)
[image]
As you can see, the comparator is mostly outputting +10V all the time, doesn't really have enough time to go all the way to -10.

This is the sort of activity I'm seeing on START_OF_LOOP (CH2 blue), so input to the shift register
[image]

@TomWhitwell does this seem like normal operation to you?

QrchackOfficial, 2020-07-20:
Managed to source a known good CD4016, checked the noise+comparator circuit on my scope, works as advertised - from V- through random bits in the middle into V+ at the other end. Still getting improper output, most notably the sequences are of different length than indicated by the length switch. Eg. I'm getting 12 steps with the switch at 8. I have confirmed all the wiring from END_OF_LOOP to the individual bits outputs from the corresponding shift registers. All good there.

jstamant, 2020-07-22:
@QrchackOfficial my TM works flawlessly or erroneously depending on the clock source. Using my B&K Precision function generator as a clock source, my TM works flawlessly. Using a Make Noise MATHS however, my TM exhibits the above-mentioned behavior - dropping bits. Good job so far with the troubleshooting, but in my case I wasn't really suspecting the noise generator too much. I am still concerned with impulse noise messing with the shift register's operation, but I don't know how to test this properly.

If we isolate the shift register from the remainder of the synth modules, strip-back as much of the circuitry as possible, and STILL manage to recreate the problem, we could get engineering support from the manufacturer. This could also be worth sharing in electrical engineering forums or subreddits to see if anyone else has seen this issue.

I think I posted this issue in 2018 to get the information out there, and was planning on coming back to flesh it out a bit more. Looks like I never came back, or lost interest! Hopefully this additional info helps. I'll see if I can make some time to set-up my modular again and test this.

QrchackOfficial, 2020-07-22:
@jstamant this might be a clue, I've been using a Doepfer A-110-1 on square and Vector LFO from Hemisphere Suite, also on square. Actually, I've found it to trigger on even just a triangle LFO. That said, I think I've been hearing the shift register trigger on both rising and falling edge of the clock, resulting in two notes per clock pulse. And I think it wasn't consistent.  I'll check the clock then, I think on the scope I got a missing pulse once but will have to verify.

I was testing noise first since this seems to be no 1 issue on muff and I thought this could be it. Went through pretty much the whole circuit, spent a bunch of time testing pretty much the PCB and connectors just to be sure, but also got to read the datasheets. I've even spent good 30 minutes drawing the whole loop control circuit, including analysis of the comparator, and calculating bits coming at START_OF_LOOP for all possible combinations of comparator output and END_OF_LOOP :D So, I got a decent look into how it's supposed to work, and verified with my scope that the noise section is all good.

I have one more suspect now, too. I built this one from PCB+Panel, and have used ceramic caps for a lot of places where the kit includes plastic film. Asked Tom about this before building, he said it should be fine just like I thought. But this might be of importance for triggering, and I'm not sure if I can trust chinese ceramics that much. Will investigate further.

QrchackOfficial, 2020-09-27:
Just went back to testing. Set up a quick 555 timer clock source and powered the whole thing with my bench PSU. If this ain't a good enough clock (at CLOCK_FRONT, so after the TL074/U1), I don't know what is:

[image]
[image]
[image]

And it shows the same behavior - a single 1 running across, skipping LED 2 and going right to LED 3, after reaching LED 8 there are 8 counts for the remainder of the shift register, and the cycle continues. This is with change set to far right and register length set to max. If I input five 1s in a row manually, after several counts they disintegrate into the same single 1 as at the beginning, and they never come back.

QrchackOfficial, 2020-09-27:
The datasheet for CD4015 lists maximum rise/fall time of 6us at 10V and a minimum clock pulse width of 80ns at 10V. The clock I have measures about 10ns and is almost 150ms wide, so it can't be the clock.

QrchackOfficial, 2021-07-02:
Update on the issue. I've recently set up a handy jig, with two lab PSUs, a clock source and everything needed to troubleshoot this module. I have also used software (eagleview) to convert the PCB files to a boardview file, as seen in action by board-level repair technicians. Openboardview, together with EAGLE image export of the board preview gives me this:

[image]

with the ability to search by netname or component reference. Nice stuff, especially when you need to find a place that makes measurements easier. Should this be of help to anyone for troubleshooting, I can provide the files.

I've gone through the schematics and ruled out AND_1, AND_2 (both 4081s, only used for GATES output), BUFFER1, BUFFER2 (4050 only used for PULSES output), as well as SHIFTREG_EXT (which provides steps 9-16). I've set my loop length to 8 and provided a known-good clock with very good rise/fall times, well below the maximum specified in CD4015 datasheet. I still get the same result. On one test I got the following behavior:

clock 1 rising edge - LED 1 (from top left as seen by the user, so B8_FRONT) turns on
then immediately on falling edge LED 2 turns on
clock 2 rising edge, LED 4 turns on (3 seems to be completely skipped)
clock 3 - both 5 and 6 turn on
clock 4 - both 6 and 7 turn on
clock 5 - both 7 and 8 turn on

This is all with loop control pot set all the way right, and I'm getting about 10.5V constant from pin 7 on the TL072. I've tested the CD4016 on a breadboard and it seems to be _super_ sensitive to any input that is not ground (as expected with a CMOS chip), so as a precaution I've bodged a 100k resistor from pin 13 (control input) of it to ground (the ferrite nearby is a good place to hook it up), as a pulldown to ground. This also has the side effect of creating a voltage divider that reduces the incoming control signal, but I've confirmed it's right at the +/- 5V mark, so perfect for a logic chip. In the original design it's just a comparator that shoots the full +/-12V. There is some progress with this mod - now if the loop control is all the way to the right, and I set a loop length of 8, I get 8 LEDs lighting up one by one on every clock rising edge, and then they turn off in pairs (1+2, 3+4, 5+6, 7+8). Then the whole thing repeats. No bits get dropped.

Still, I can force the whole thing to all zeros (with the write switch), but then it appears as if a constant logic 1 is present at the input of the shift register, so it's back to 8 leds one by one etc, as described above.

I'll experiment with it some more and report back if I find a full solution to the problem.

QrchackOfficial, 2021-07-03:
Spent some more time with it. Here's a rewrite of the schematic for anyone struggling to understand the logic done using the CD4016 bilateral switch IC. This assumes LENGTH is set to 8 for a one-to-one relation between the bits and the LEDs.

[image]

QrchackOfficial, 2021-07-03:
Everything so far points to the shift register itself. Can't get it to work reliably on a breadboard despite pulling unused inputs low. Will source another one and update.