TomWhitwell/TuringMachine #90: expander pulses 5 always pulsing, just like the clock source

indraperkasa, 2018-03-24:
Hi, i've built my turing machine and all the expanders few months ago, and its been working just fine. until recently i just noticed my pulse expander, on the pulse 5 is ticking exactly like the clock input on the main turing machine. from my understanding, it shouldn't do that (or am i wrong?).
the rest of the pulse working normally, have the same LED movement just like the LED from the main turing machine. i have all expander connected to the turing machine. 
i try to switch, connecting the pulse expander to the gate connector (instead of pulse connector) on the back of the turing machine, and it work normal, the LED on turing machine reflected the same movement like in the pulses expander. so i suspected, there's a problem in the pulses connector at the turing machine, but i don't know what to do about it.
any idea what is broken? or maybe should i replace any IC?

thank you.

jstamant, 2018-03-24:
@indraperkasa Very good troubleshooting on your part. From what you're telling us, I'm comfortable in saying that there's no problem with the pulses connector.

Here's what I'm suspecting:
- Pulse 5 is shorted to 12V
- One of your 4081 quad AND gate ICs are defective.
- One of your 4015 shift register ICs are defective

Is the 5th LED on the Turing Machine always lit?

Try swapping the two 4081 ICs on your Turing Machine to see if the problem moves from pulse 5 to pulse 3. If it does, the 4081 that had pulse 5 on it is defective.

You can also try swapping the two 4015 ICs to see if that clears the problem.

If after you've swapped the two 4081 ICs, and the problem still registers on pulse 5, you'll have to start checking some voltages and continuity. First check the continuity on all the pins of the 4081 quad AND gate chip that carries pulse 5. Make sure that each pin on the chip has continuity to where they're supposed to go, and not connected to +/-12V or ground if they're not supposed to. Pin 8 on the 2nd AND gate IC (closest to the board edge) should have continuity to pin 4 on LINK3, pin 13 of the main 4015 shift register IC (closest to board corner), and should NOT have continuity to the +12V rail.

If all that still doesn't shed some light on your issue, check the voltage on pulse 5's AND gate inputs. Sounds like pin 8 with the B4_BACK signal is stuck high, and is just passing through the clock.

indraperkasa, 2018-03-25:
@jstamant hi Justin, thank you so much for the quick response. 
the 5th LED on turing machine doesn't always lit, only on the pulses expander.
another thing i noticed, if i hold down the toggle write switch, all the LED goes off, except the LED 5 on the pulses expander, which pulsing at the same rate with LED clock input at the turing machine.

i've tried swaping the the two 4081 IC, didn't make any different. same result with the swapping of the 4015IC.
forgive me, since i'm a very newbie, don't really know about this electronical stuff, just got into soldering when i start to play modular a few months ago. i don't really know how to check voltages and continuity. if you don't mind, maybe can you guide me? i have a digital multimeter. i've been youtubing on how to do it, just not really sure if i was doing the right thing.
let say, if the Pin 8  on the 2nd AND gate IC4081 doesn't have continuity, what should i do next?

and about your last suggestion, can you guide me how to do it? checking the voltage on pulse 5's AND gate input (i don't know which one..). and if the pin 8 with the B4_BACK is stuck high, what should i do next?

thank you.

jstamant, 2018-03-25:
No worries, @indraperkasa.

You'll need to refer to the [schematic](https://github.com/TomWhitwell/TuringMachine/blob/master/Collateral/PDF_Schematics%20/TuringMachine2_Schematic_May2016.pdf) for the remainder of your troubleshooting. Also, downloading Autodesk's EAGLE will let you open the `*.brd` files located [here](https://github.com/TomWhitwell/TuringMachine/tree/master/Hardware) to get an x-ray view of the board, which you might find helpful for troubleshooting.

When you're checking continuity or checking resistance between two points, you need to have the board powered off. With your multimeter set to measure resistance, touch the two probe ends together, and make sure the multimeter reads close to 0 ohms. Next, take a spare resistor and measure its resistance. If the multimeter is reading the right value, then you know your multimeter is working, and you can proceed to checking continuity and voltages.

Again with your multimeter set to measure resistance, touch one probe to pin 8 of 4081 IC closest to the board edge, and touch the other probe to one of the +12V pins on the power header (opposite end of where the red strip goes). The multimeter should read "OL" or something similar indicating that the two points are not connected. Then do the other resistance/continuity checks with the other pins I mentioned. i.e. from pin 8 to LINK3's pin 4, and from pin 8 to pin 13 of the 4015 IC. All these pins are connected, so they should read something close to 0 ohms on your multimeter. Sometimes you have to press the probes hard to penetrate a thin layer of oxidization and/or flux to get a reading at all.

After you've done that, you can check voltages with the board assembled and powered on. Keep the front panel off the Turing Machine until you've confirmed that it's working. This time, to check that your multimeter reads voltages correctly, measure the voltages at the power header, because you *know* it carries +/-12V and 0V, or your module wouldn't power on. Placing the black lead on a GND pin, and the red lead on a + pin should read 12V. Then trying the red lead on a - pin should read -12V. Good, now you've confirmed that you multimeter is fine. Keep the black lead on the GND pin, and place the red lead on pin 8 of the 4081IC that's closest to the board edge to check the voltage on that pin. It should read 0V if there's no pulse on LED 5 on the Turing Machine. It should read 12V if there IS a pulse on LED 5 on the Turing Machine. From what you're telling us, it sounds like the voltage is *always* 12V, but we need to confirm that by measuring it.

indraperkasa, 2018-03-26:
@jstamant hi Justin, 
thank you so much for writing this step by step, really appreciated it.
i did some test based on your guide.

the results are:
pin 8 of 4081IC closest to the board edge to +12v = approximately 6 mega ohm
pin 8 of 4081IC closest to the board edge to LINK3's pin 4 = approx. 18 mega ohm
pin 8 of 4081IC closest to the board edge to pin 13 of 4015IC = approx. 6 mega ohm

and with the board assembled and powered on,
pin 8 of 4081IC closest to the board edge with ground pin, reads 0v when LED 5 is off on turing machine, also reads 0v when the LED 5 is on.

indraperkasa, 2018-03-26:
[image]
just to confirm if i did test the right pin, here i attach the photo, i put the red circle for the pins that i think the one that you refer to..i hope its the right

jstamant, 2018-04-01:
@indraperkasa How confident are you with your use of the multimeter? The pins you outlined in your picture are the right ones, so you were taking measurements on the correct pins. I was hoping on getting different results than you indicated, though.

I was expecting pin 8 of the AND_2 chip to be either shorted to the +12V rail, or at least showing a voltage >+8V at all times (the 4081 IC closest to the board edge is named AND_2, the other 4081 IC I don't care about is named AND_1). You indicated that the pin shows a crazy-high resistance to +12V, meaning they're not connected, and that the pin is always showing 0V while running. If that pin shows 0V, then LED 5 on the Turing Machine should be off. Whenever it shows a voltage, LED 5 on the Turing Machine should be on.

Then, pin 8 of the AND_2 chip is supposed to be connected to LINK3's pin 4 and pin 13 of the main 4015 shift register, i.e. showing ~0 ohms. According to your measurements, in the mega-ohms range, they're not connected.

Let me know what you think of these results.

indraperkasa, 2018-04-02:
hi Justin,
i'm pretty confident with the result form the multimeter, i've checked it several times and the result is pretty much the same.
however there's bad news, i might damaged the module during the testing voltage procedure. i accidentally put the black lead from the multimeter (slip from my finger) in the -12v pin, and it causes spark a few times. 
now, when i try to input a clock, the LED on turing machine doesn't even do random pattern anymore. it still receive clock pulse, but when i hold down the write toggle, after all the LED off, it remains off all the time. also if i hold up the write toggle for once, one LED show up, but that's it, only that one LED run across the all the 8 LED, looping.
i've been thinking, maybe just gonna let this one go..

jstamant, 2018-04-02:
Ok, no problem: a little spark is nothing we can't fix.

Where did you short the two pins? On the power header? Or was it one of the ICs? If you shorted on the power header, the damage would sooner be to your power module. If you shorted on an IC pin, that IC could be blown.

Did the spark appear on your test lead, or was it a component that sparked? Did you smell smoke when this happened? Magic smoke is the telltale sign of a failed component.

As for measuring the voltage on LED 5, that suggestion is still applicable. You can still check the voltage on that. It sounds like you can also troubleshoot the original issue with pulse 5 regardless of this other issue.

indraperkasa, 2018-04-02:
the sparks was on the power header. and it did make my power module to stutter, the +12v led lights blinking rapidly for a moment, and then back to normal. 
thanks so much for the suggestion, i'm gonna try it again.

indraperkasa, 2019-06-13:
hi, just wanna say i finally solved the problem after abandoning it for more than a year. turns out the pin 8 iC 4081 (close to the edge) didn't connected to where it supposed to, turns out there's a scratch in the pcb, that broken the Via connection. so i solder a wire from pin 8 IC 4081 (close to edge) to the pin 5 IC 4050 (close to the edge). and now its working.
just wanna say thanks for your help, i learnt alot from debugging this, i feel a little bit confidence now on troubleshooting. and i'm getting more familiar with eagle file, such a convenience way.
thank you.

closed 2019-06-13 by indraperkasa

TomWhitwell, 2019-06-13:
So glad to hear it worked, @indraperkasa