TomWhitwell/TuringMachine #91: Turingmachine does not create bits

foenzie, 2018-03-28:
Hi guys,

My turingmachine has been working for about a few months. trouble during my build was a short around the powerheader, but this was fixed and after calibration the unit worked. 

After a while I noticed the lockedloop started dropping bits. So after reading I replaced the 4015 chips. This did not solve the problem. 

After a few more weeks of being useless in my modular, I noticed the turingmachine did not produce any bits at all. With the write switch moved up it still generates only 1 bits and imediatly drops the bits out of the register. 

Luckly enough I have a friend with a working turingmachine. I started swapping different chips from his turingmachine into mine turingmachine. Did not solve the issue. All chips have been replaced and the problem persists. 

Anyideas?

jstamant, 2018-03-28:
Can you explain how your write switch works a bit more, and how the bits from the write switch behave? When do the bits dissapear? Do the bits do a full loop, then drop; or do they dissapear halfway-through or on a specific bit?

TomWhitwell, 2018-03-29:
Sorry to hear this - one thing to try if you have a friend with a working Turing is swapping the front and back boards, to identify where the problem lies.

foenzie, 2018-03-29:
@jstamant hi thank you for the fast reply. The bits are created holding the write switch up. I also get a pulse on every bit. The bit makes it all the way to the right, and then never returns.

TomWhitwell, 2018-03-29:
First of all thanks @jstamant for your amazing support in this and other threads! 

If the bits are failing to loop round, and the 4015 and 4016 chips are definitely OK, then some things to check for: 
- bad solder joint around the rotary switch - it has to get through the switch to loop. 
- Solder joints on the 4015 and 4016 chips - the chips might be fine, but if a connection is bad, or there is a short, then it won't work. 
- A bad connection or something around the main Change pot - I spent about a year debugging a Rampage only to find the main pot was somehow dead - if you've had a shortcircuit around the power in the past, it might have fried something? 
- Just checking that all the chips are properly seated can be worth doing. 

Good luck and thanks again @jstamant