TomWhitwell/TuringMachine #92: Turing Machine Dropping Notes + LED 2 out

BlipSqueek, 2018-04-01:
I have 2 issues with my build. I have built many modules so I'm not new to this, but I am also not someone who is good at troubleshooting circuits.
1) LED2 is out, but I do appear to be getting notes from 2
2) The turing machine will drop notes. If I have the main pot at around 3-4 o-clock it will cut out periodically, It always comes back within a second or 2 but it drops out pretty consistently. Maybe once every 5-10 seconds.
Here is a zip of the audio file ( won't let me upload wav files)

[TM Drop Outs.zip](https://github.com/TomWhitwell/TuringMachine/files/1866300/TM.Drop.Outs.zip)

jstamant, 2018-04-01:
If LED 2 is always off, the LED could be soldered on backwards. But don't remove it just yet. Try checking the voltage across that LED's leads when you suspect it's supposed to be lit. You should see ~2.2V when the LED is on. When the LED is supposed to be off, you should see 0V across the leads.

Or trying something else: measuring voltage from the + lead of the LED to ground, you'll see ~12V when it's supposed to be on, and 0V when it's supposed to be off. If you have it soldered backwards, you'll still see the same 12V/0V voltages.

As for your 2nd issue, how do you have your Turing Machine patched when you see this problem? Have you successfully calibrated your Turing Machine? And just for speculation, the Turing Machine doesn't produce notes, it produces random voltages. Your audio file sounds like you're clocking the TM at an audio frequency, and running the OUT jack right into your recording device or speaker, so I would expect it to behave as it sounds in your recording. Occasionally the TM might randomly clock in a set of all 0's, so there would be no change in output voltage i.e. no audio waveform to hear.

BlipSqueek, 2018-04-02:
Thank you for the help.
It appears that the only thing wrong then is the LED. It calibrates fine.
I tested the voltages. On the LED's ( + to -) I'm getting 1.69V on the working ones and nothing on the one that is out. Going to ground I get the expected 12V on all LED's. Did I put the LED in backwards? I triple checked and was sure they were all correct.

closed 2018-04-02 by BlipSqueek