TomWhitwell/TuringMachine #93: Very hard to hit the "random" spot on the lock knob

omidderham, 2018-04-01:
Hi,

I have a very hard time setting the lock knob to a position where i get something random. What can this be an indication off ?

Regards,
Olivier.

jstamant, 2018-04-02:
Have you calibrated your Turing Machine?

omidderham, 2018-04-02:
Yes i did. What happened was that with the trimmer is fully clockwise, i have a steady tone. As the steady tone looked to be the desired end result in the vid i thought nothing more of it and thought i was done. Now i notice i don't really have any randomness on it.

One other thing i'm noticing is that the noise volume is rather low and i get the impression that might be related.

jstamant, 2018-04-02:
You're definitely on the right track. Check that all the components/pins in the noise generator circuit have continuity to the components they're supposed to be connected to. You can also check continuity between adjacent pins or pads that are close to eachother to ensure there are no unintended shorts.