TomWhitwell/TuringMachine #94: Advice on troubleshooting

Steckdose23, 2018-04-12:
Hello,
I have a Turing Machine MK2 where the 5th & 6th LEDs (from the left looking at the front panel) do not light when a sequence is running. The 6th on both voltages and pulses does work however. Both LEDs are connected with the correct polarity and I can get a light from both with a multimeter. The fifth seems to have continuity at several points on the board (which seems odd as the other LEDs don't). 

Where do I start with the troubleshooting? 

Thanks in advance!

Lewis

TomWhitwell, 2018-04-12:
HI, sorry to hear that. 
If you have a look at the schematic, you can see where the LEDs are connected: B6_Front and B5_Front - it might be that somewhere one of those B6 or B5 signals is shorted to ground. 

If you (or a friend) have another Turing, it's easy to swap the front and back panels, to quickly isolate which board has the problem. 
Good luck!

Steckdose23, 2018-04-12:
Hi Tom, thanks for replying. 
So it sounds like a short to ground? What if I was to disconnect the led from board entirely and use jumpers to connect it to the correct places? I did make a pretty bad mistake with the link 3 header on the back board which trashed the board so had to hack a stripboard substitute that seems to work apart from these led 5&6 issues...the downside is that the boards are inseparable. I had to discover how to read the schematics and look into the pin layouts of the ICs, it was very informative and part of the reason I decided to go diy for most of my modules. 

Lewis

JeffGreenlee42, 2018-04-18:
Just a comment:   I had the same problem (one of 2.. the other I haven't solved yet :(  )..  I ran continuity checks with my multimeter and discovered that their resistors were not properly connecting to ground.  I replaced both resistors.  One solved the problem,  The other resistor was still not getting a connection to the negative side of it's corresponding LED... so I just soldered in a jumper wire . ugly looking... but problem solved.   One thing about PCB's that I really like - that doesn't get mentioned very often -  is that if you hold the PCB up to a bright lamp you can can follow exactly where the traces lead... and when you start to get familiar with this .. you can tell which traces lead to ground vs. what leads to voltage.   If you can't see where a particular connection leads to, just turn the PCB over and you can often  find a trace there.  If you see large dark masses where the light is occluded,  those are usually a substrate of  ground.     I find this really helps me make sense of the physical layout and I find it helps makes the logical Schematic layout correspond to the physical layout.

Steckdose23, 2018-04-18:
Thanks for the tip Jeff! 
The covering on these boards seems too opaque to see though but I'll give it a go.

I have managed to fix the bit 3/pulse 6 non lighting led with a jumper (the fact it was sending to the expansions suggested to me it was just the led area that was affected), but I'm still stuck with the b4/p5 not working on anything so I think that's caused elsewhere and at the moment I'm not skilled enough to right now to follow where it's coming from. I did notice that the LEDs had continuity in ways I didn't expect, but again lack of experience is probably to blame... So I've got another kit on order to compare with  and hopefully figure out a fix. Or not!

Steckdose23, 2018-04-18:
I will try that as soon as I can, thanks for the tip!
Will that tell me whether or not that Bit is firing?

JeffGreenlee42, 2018-04-18:
Yes, it will indicate that your chips are working and signals are getting
to the LEDS. If they are getting signals, then the challenge is finding out
why there is no path between ground and the negative leg of the LED.