TomWhitwell/TuringMachine #95: No lock when length set above 8 steps

Shayshez, 2018-05-07:
Just finished to build Turing machine mk 2 and it's almost work perfectly...
when I set the length between 2-8 steps everything work as it should 
but if I set the length above 8 (12/16) there is no lock, instead if I rotate the knob fully clockwise it set all the steps on low value and if I set it fully counterclockwise it set all the steps on high value (like the operation of the write switch).
I built it from mix of 2 kits (from thonk) the beck pcb is rev 5 and the front pcb is rev 4
is there anything I can do to fix this issue?

ACWIZ, 2018-08-26:
I'm having the same problem that i've been trying to fix for over a year now. Mine does not lock at 6 or above, the notes then leave the sequence one at a time every time it loops.

deepiain, 2018-08-26:
Well try that TL074 next to the DAC.. or the socket it goes into. 

good luck man

ACWIZ, 2018-08-27:
Thanks for the pointing me in that direction, i'll reflow the joints there again and see if swapping the ic with a new one will help.

ACWIZ, 2018-09-05:
swapped the TL074 with a new one and it's working fine now. Thanks Deepiain!

deepiain, 2018-09-05:
That’s great to hear man! :)