TomWhitwell/TuringMachine #96: Calibrating issues. Noise output problems

Cseley, 2018-06-11:
Hi,
Friend of mine put together a TM for me but he is not familiar with the modular world an I am not familiar with electric circuits, so now we try to figure out together what went wrong. Great :)
1. The most obvious symptom is there is no white noise at the noise output but I can here some clicks in the tempo of the clock input. The calibration pot acts like an volume control to this "noise" (click)  output.
2. When I tried the calibration process, regardless of the calibrating pot position, there was no glitching inbetween 12-5 oclock position. It only happened from the 7-12 pos.
3. There is not any randomness introduced to the pattern between 12-5 oclock, it only lockes. Than suddenly playing fully random around 11-12 oclock.

TomWhitwell, 2018-06-11:
Hi, sorry to hear you're having problems. 

I can't get the video to play, but if there is no noise at the noise output, the module will definitely not work properly. 

Take a look at this thread to see if you can find any useful tips

https://github.com/TomWhitwell/TuringMachine/issues/69

Cseley, 2018-06-12:
After all I found a mistake. The 2N3904 and TL431 transistors where mixed up.  Now I have to find someone to switch them to there correct location. Could such thing cause such problems ?

TomWhitwell, 2018-06-12:
That’s good news - if it was me I’d find new components for both - they’re easier to remove if you don’t mind killing the old one + means you know they’re good.

closed 2018-06-12 by TomWhitwell

Thonk-Steve, 2018-06-12:
I'm pretty sure that when other people made this mistake that it was ok to
switch them (i.e. no other parts were damaged due to it), BUT I'd recommend
using new parts for those two pieces rather than swapping them over as they
are cheap, are far more likely to get damaged during desoldering and it'll
simplify any further debugging. It's also better to destroy them to
desolder (i.e. cut them so you only need to desolder one pin at a time) as
you then don't risk the PCB itself getting damaged.