TomWhitwell/TuringMachine #97: Turing machine not random. Also, no noise

RogePiltoney, 2018-07-11:
Hello - this is maybe my 3rd or 4th DIY build, and I am having trouble with getting the Turing Machine Mkii to work. Calibration worked fine, the pulse and main outputs work fine, the CV and clock inputs work fine, and the length and scale knobs, as well as the write switch, also all work fine. What's wrong is the big knob and the noise output. 

The big knob is not random no matter where it is - it is always locked. Between 12 and 5 oclock, it locks into a sequence whose length is equal to the length knob; between and including 7 and 12 oclock, the sequence is also locked, but its length is twice the length knob.
However, I noticed that if i continuously and constantly move the big knob back and forth (or put an LFO into the CV input), I can manage to get a random sequence. 

Also, the noise output does not seem to work. 

I've already tried re-soldering everything, and have thoroughly cleaned all sides of the PCB with 91% isopropyl alcohol.

Any advice would be greatly appreciated! Would pictures of the PCB be helpful?

Thonk-Steve, 2018-07-11:
Hey,  yeah pictures would really help. It sounds like the problem is
isolated in your noise circuit. (video wouldn't hurt either to confirm this
though)

The schematics are here -
https://github.com/TomWhitwell/TuringMachine/blob/master/Collateral/PDF_Schematics%20/TuringMachine2_Schematic_May2016.pdf

If you look at page 1, in the middle at the bottom you can see the 'noise
generator' that starts with the 'T1' transistor and ends with NOISE-OUT.

Check the soldering again of all components in that section, it's
definitely a 2N3904 in for T1 and not one of the other similarly sizes
parts? The TL074 is the right way round.

RogePiltoney, 2018-07-12:
Hi, thanks for getting back to me! I tried switching the 2n3904 with a spare one I had lying around. I am now noticing an extremely quiet output from the noise output, but for all I know that was there before (i hadnt tried turning the volume of my mixer all the way up). The sequence is still not random.
Is there a way I can check the integrity of the noise circuit? I'm not super experienced with electronics, but do have a multimeter. I'll try resoldering everything in the noise section. 

here are some pictures.

[image]
[image]
[image]
[image]

THANKS!

RogePiltoney, 2018-07-12:
I tried resoldering everything and still it is not random.

Thonk-Steve, 2018-07-12:
You could try replacing the TL074 next, it may have got damaged. You said
the noise level is very quiet, does that very low level change though when
you listen to it directly and move the trimmer? Do you have a digital
multimeter?

RogePiltoney, 2018-07-13:
I tried moving the trimmer and low and behold, out comes the noise! The sequence is also now fully random, but has some weird behavior. 

Sometimes when fully clockwise it will loop a sequence that, say, every third time is different. It's still a loop, but just not consistent. See second video (and towards the end of the first). This only happens occasionally. 

It also often will get stuck with all the lights on or off and sometimes if it gets stuck too long, I'll have to press the write switch to fix it. See video.

When the length is 2, no matter what I do, the locked sequence is always the same two notes. See video.

Finally, the sequences are fully random at twelve, but seem to lock into a sequence around 9 and 3 o'clock, as opposed to getting gradually less and less random until fully counterclockwise or clockwise. See video.

https://drive.google.com/file/d/18iQcgIdChUqeMJoqKaQLI3IwFmdtNQ2v/view?usp=sharing

https://drive.google.com/file/d/1f2GlG3ZLt5O43lXu0qthMYneCF40VKgn/view?usp=sharing

Thonk-Steve, 2018-07-13:
> Q2v/view?usp=sharing
> Kgn/view?usp=sharing

Did you re-run the calibration again and re-trim?

RogePiltoney, 2018-07-16:
Yes.