TomWhitwell/TuringMachine #100: Disambiguation backfired

robertsyrett, 2018-08-09:
I accidentally soldered the TL431 in where the 2N3904 should be and am really struggling to desolder.  I think you should edit the passage on page 13 so that TL 431 isn't in Bold.  I know it's my mistake, but I was looking at my iPad from a few feet away and got mixed up.

Honestly, I am sad and unsure if I will finish now.

robertsyrett, 2018-08-11:
I guess this is a dead forum.

TomWhitwell, 2018-08-11:
Hi Robert, Sorry to hear you had trouble - I wasn't sure what question you were asking - thanks for the feedback on the build documentation.

closed 2018-08-11 by TomWhitwell

robertsyrett, 2018-08-13:
Happy ending, I persisted with watching hours (literally 10s of minutes) of desoldering videos and was able to desolder the transistor and there was enough lead left to get everything into the right spot.  Everything works!  Well, except for the led on the pulses output, but at this point that seems like small potatoes.

In fact, I am looking forward to trying out some of your other kits. :)

Again, I hope you can see the benefits of only BOLD facing the correct parts and not the incorrect parts.  Ty for getting back to me.