TomWhitwell/TuringMachine #101: No randomness

eliasbagley, 2018-08-16:
Hey guys!

I'm having some issues with the main randomness and length knob not affecting the behavior of the module at all. Hopefully you guys can point me to the part of the circuit that might explain the issues I'm having.

The issue is that no matter how I turn the main randomness knob or the length pot, it doesn't affect the sequence at all. I _can_ get values out of the output when I use the write switch, however. The write switch seems to be "sticky", in the sense that if all the lights are off, and I insert a single bit with the write switch and then release it back to the center off position, it will continue to insert bits from the left, as if I were holding the write switch up. No 0 bits ever get inserted again, until I manually write it in by flipping the write switch down. The same "sticky" behavior happens when I write a 0 - 0's fill in from the left as if I were holding the write switch down, even when I'm not. I don't think this is an actual issue with the switch though - I'm guessing that the part of the circuit that is doing the randomness and flipping part isn't working as it should, but I don't really understand that part of the schematic to debug it further.

I'm unable in any scenario to get bits to "wrap around" from the right to the left. Which part of the circuit controls this randomness and wrap around behavior? I'm guessing the issue is in there somewhere.

Here's what I think I know from the parts of the circuit that are behaving correctly -
1) Clock circuit works. The LED blinks and when I use the write switch, the top LEDS blink along with the clock.
2) Noise circuit works. I hear a strong white noise sound from the noise output, which is attenuated down to inaudible when I adjust the trimmer pot with a screwdriver.
3) Write switch works. Flipping it up writes a value into the left bit, although it's "sticky" - I don't think this is an issue with the switch though.
4) Scale pot works
4) DAC seems to work - the pitch of the output does change when I manually write bits with the write switch, and corresponds to the binary value indicated by the top LEDs.
5) Pulses LED blinks when there is activity on the top LEDS from manually writing bits

Any ideas? I'm planning on poking around the circuit a bit more to try to rule out any broken traces. If that doesn't work, which components are likely at fault? I might order some replacement components just to rule that out. 

I've already reflown all the solder joints, and they look pretty decent to my eye, verified that no IC pins are folded back on each other, swapped like ICs around to see if that changed anything, and verified that there's no shorts between +12, -12, and GND. Problem persists in all these cases.

I'm happy to post some high res photos of the boards or a video of the behavior, if that helps.
Thanks for your help!

Thonk-Steve, 2018-08-16:
>    5. DAC seems to work - the pitch of the output does change when I
>    6. Pulses LED blinks when there is activity on the top LEDS from

I'm on holiday at the moment so can only provide brief feedback, have you
used a continuity checker on your multimeter to make sure that all the
mating pin headers between the two boards are connected properly and that
no pins are accidentally bridged? It sounds possibly like the back board is
working fine but the controls aren't getting to it.

Steve

Thonk-Steve, 2018-08-16:
Oh and when and where did you buy the PCBs? which revs are the front and
back boards?

eliasbagley, 2018-08-16:
Thanks for the quick response! I'll check the mating pins today. I bought PCBs from Thonk - I don't have the PCB in front of me right now but I believe it said Version 5 - August 2017.

Thonk-Steve, 2018-08-16:
ok let us know, the revision number will likely be different on each board

eliasbagley, 2018-08-31:
I did some basic tests by pulling out the comparator circuit's op amp and 4016 and wiring them up to a simple circuit. Seemed like the 4016 wasn't doing anything, so I ordered some replacements. I'll report back to let you guys know if this solved the issue.

eliasbagley, 2019-01-02:
I figured out my issue - I burnt out one of the traces on R16, which sends the noise control signal into the start of the shift register, so it wasn't getting any source of randomness. I fixed this with a simple jumper wire. It's working beautifully now :)

closed 2019-01-02 by eliasbagley