TomWhitwell/TuringMachine #102: Double Triggering

Triscus, 2018-08-26:
If I feed the trigger from the Pulse Jack or the Pulses Expander into an Erica Synths T->G (Trigger to Gate) I noticed the trigger is received a second time at the moment the next step is triggerd . So when using 3 Steps and only one step is HIGH, the T->G is triggered on Step 1+2. 

The TM was built by me, the Kit was bought from Thonk. Other modules working fine with the TM and T->G. VOLTS Expaner is used too.

I would think it's a kind of incompatibility between the modules, but I'm struggling to find an explanation. 

Any Ideas?

jstenroos, 2018-08-26:
Hi,

I asked about this in the Muff forum once without answer. Then I showed the schematics to my friend who is a 
hardware guy and he spotted the problem in the design. It’s a race condition in the “AND"-operation between 
rising clock edge and falling shift register edge, resulting a very short spike in that first silent beat.

Here is a picture that explains it visually:

https://www.dropbox.com/s/40krqnwt31fsa79/Turing%20Machine%20shift%20register%20problem.png?dl=0 <https://www.dropbox.com/s/40krqnwt31fsa79/Turing%20Machine%20shift%20register%20problem.png?dl=0>

I was able to verify that spike with an oscilloscope.

In my rig the problem occurred with Maths trigger input and Radio Music reset input. I was able to fix it by soldering
small caps into the each pulse output and the main pulse output of the Turing Machine itself. The caps act as mild
low pass filters that do not remove the spike completely but lowers it enough so that the other modules do not react to it anymore.
Can’t remember the actual cap value but I will check it if necessary.

The real solution would be to delay the AND operation just slightly to allow the registers to settle in their new values,
but that would require redesigning the module and it’s likely not something you could patch in afterwards.

-Jouni

Triscus, 2018-08-26:
Hi Jouni,

thank you very much for the explanation. I just ordered the Radio Music
Module and planning to get Math too, so it would be very helpful if you
could find out the cap values.

Kind regards

jstenroos, 2018-08-26:
They are 470pf (code 471), small ceramics. Here is a picture:

https://www.dropbox.com/s/r5cldhwedo9m4d7/IMG_1140.JPG?dl=0 <https://www.dropbox.com/s/r5cldhwedo9m4d7/IMG_1140.JPG?dl=0>

The value was chosen mostly by guessing. :)

-Jouni

Triscus, 2018-08-26:
Thank you,  I'll try it next week when I got a soldering iron in reach.

RichTea909, 2021-04-18:
Any chance you can upload the picture of the cap placement to stop the double tigger again jstenroos?  Thanks

jstenroos, 2021-04-19:
I have picture only of the pulses module, here it is. The caps are simply between the signal and ground on the outputs. 

Hope this helps,
Jouni

> On 18. Apr 2021, at 23.29, RichTea909 ***@***.***> wrote:
> 
> 
> Any chance you can upload the picture of the cap placement to stop the double tigger again jstenroos? Thanks
> 
> —
>

RichTea909, 2021-04-19:
Thanks jouni